PX4/PX4-Autopilot #20009: Pixhawk 6X microRTPS build(make px4_fmu-v6x_rtps)

kh-0306, 2022-08-04:
Hi,
I'm using Pixhawk 6x

I want to communicate with Pixhawk 6X and Jetson NX via ros2-based microRTPS.
So, modifying the latest stable v1.13.0 source code.

Using the code of the path below, I created a build command that allows rtps to be used in 6X.
(boards\px4\fmu-v5x\rtps.px4board)

I checked that the build is not working due to the memory overflow problem.
So, memory overflow was resolved by deleting other unused modules and MicroDDS = n.

Although ros2 communication was successful through the Lan port between NX and 6X, 
a recent rebuild found a problem with the lan port not working properly.

Has anyone experienced Lan port of 5X or 6X not working properly like me? Please give us your opinion

Also, I wonder if the build command related to microRTPS of 6X will be officially supported in stable v1.13.1

maxpolzin, 2022-08-25:
Dear kh-0306, 
I'd like to use microRTPS/microDDS on my Pixhawk 6X. 

Could you let me know, how you created the px4_fmu-v6x_rtps target? 
Did you try micrortps over USB?

maxpolzin, 2022-08-25:
Dear @kh-0306 ,

Did you manage to get your Pixhawk 6X working with microRTPS or micro-xrce-dds?
Did you try connecting to it via USB?
Which unused models did you delete to fit the firmware with micrortps on the board?

andrejpadilha2, 2022-09-05:
I am also wondering if the next release will support microRTPS with the 6X.

kh-0306, 2022-09-20:
Dear @maxpolzin 

I'm sorry for the late reply

I ran ros2-based microds by connecting ppp to nx eth and succeeded. However, while modifying the firmware, it seems that data does not come out of the eth port of 6x at some point.

And deleted modules are like px4flow

kh-0306, 2022-09-20:
Dear @andrejpadilha2 

I also hope that RTPS firmware will be provided in 6x from the next stable version of 1.13.1.

benberry3d, 2022-10-26:
Does this commit addresses the issue (it's added to default for the 6x): https://github.com/PX4/PX4-Autopilot/commit/cea185268e133ad22418fb9990f9c19df6e30681

closed 2025-10-27 by github-actions[bot]