PX4/PX4-Autopilot #20010: About external IMU sensor integration

qaz6906, 2022-08-04:
I want to link the external IMU.

I'm publishing 400hz of acceleration and angular velocity data through uORB.

However, I am having a hard time because I do not understand the IMU source code below.

As far as I know, IMU has only 3 accelerations and 3 angular velocities, so I wonder how to input these IMU data.

dagar, 2022-08-04:
That particular code is specific to the ICM20649 where we transfer a buffer of samples (FIFO) at once and have to deal with accel being duplicated.

What's your actual IMU? If it's a new driver you can simply publish `ORB_ID(sensor_accel)` and `ORB_ID(sensor_gyro)` directly. The downstream sensors pipeline will automatically determine the output data rate, handle integration, filtering, etc.

qaz6906, 2022-08-04:
> That particular code is specific to the ICM20649 where we transfer a buffer of samples (FIFO) at once and have to deal with accel being duplicated.
>
> What's your actual IMU? If it's a new driver you [...]

The imu I have is VN-300. VN-300 has a period of up to 400 Hz.

dagar, 2022-08-04:
Oh, great. Did you see I started a VectorNav driver? https://github.com/PX4/PX4-Autopilot/pull/19261

I'd be happy to get it finished and merged, I'll try messaging you on slack to discuss.

qaz6906, 2022-08-04:
> Oh, great. Did you see I started a VectorNav driver? #19261
> 
> I'd be happy to get it finished and merged, I'll try messaging you on slack to discuss.

Thank you. I am also interested in integrating the PX4 with the expensive sensors such as VN300 and VN310. I'd appreciate it if you could send me a message.

qaz6906, 2022-08-08:
> That particular code is specific to the ICM20649 where we transfer a buffer of samples (FIFO) at once and have to deal with accel being duplicated.
>
> What's your actual IMU? If it's a new driver you [...]

I succeeded in the way you told me. Thank you.

Additionally, I am testing the source code (#19261) that you uploaded.

The board I am using for the test is Pixhawk 6X.

 However, it seems that the data is not being decoded because the VN300 setting is wrong. I would appreciate it if you could look at the picture and tell me if there is anything wrong with me.

[image]

[image]

[image]

[image]