PX4/PX4-Autopilot #20025: Holybro kakute H7 mini,  it can't save parameters without SD card

kokoory, 2022-08-09:
Hello, 

I use Holybro Kakute H7 mini uploading kakute H7. It works anyway.
But there is difference from H7 mini with H7. H7 has SD card, So, I could save the parameters.

Is there way to save parameters in flash memory for kakute h7 mini?

Thanks,

bkueng, 2022-08-09:
Hi

Yes, flash-based parameters for H7 just came in recently in https://github.com/PX4/PX4-Autopilot/pull/19838, so we can switch to that now. Do you want to take that on?

kokoory, 2022-08-09:
> Hi
> 
> Yes, flash-based parameters for H7 just came in recently in #19838, so we can switch to that now. Do you want to take that on?

Thanks, your reply. 
I want to do that. How should I do?
I compiled and uploaded but I can't save parameters.

bkueng, 2022-08-09:
You need to enable it in the code. See 2698625be972ca6 and 169722c46ff5073. The flash sector is already reserved.

kokoory, 2022-08-09:
> You need to enable it in the code. See [2698625](https://github.com/PX4/PX4-Autopilot/commit/2698625be972ca619729ac10bbbaf1ad0197dd3d) and [169722c](https://github.com/PX4/PX4-Autopilot/commit/169722c46ff5073b634770ac0d817e3acf3c5602). The flash sector is already reserved.

Thank you very much!

I check the file and code where you referenced. 2698625 is OK.
But, [https://github.com/PX4/PX4-Autopilot/commit/169722c46ff5073b634770ac0d817e3acf3c5602](url) is not present in the omnibus and file is changed.
Nevertheless, if I write down the code in rc.board_defaults, will it work?

bkueng, 2022-08-10:
> Nevertheless, if I write down the code in rc.board_defaults, will it work?

Yes it should.

bkueng, 2022-08-22:
@kokoory any update? Can you create a pull request?

kokoory, 2022-08-22:
> @kokoory any update? Can you create a pull request?

As I mentioned above, All the code is correctly located except the below.

just add below code to "boards/holybro/kakuteh7/src/board_config.h". check https://github.com/PX4/PX4-Autopilot/commit/2698625be972ca619729ac10bbbaf1ad0197dd3d

this is code to add board_config.h  -->   #define FLASH_BASED_PARAMS

But SD card warning appear in the QGC, but I don't care.

I don' familiar with GIt, So I could not create a pull request.

Thanks,

bkueng, 2022-08-23:
Alright, I'll have a look.

closed 2022-08-31 by bkueng

RENyunfan, 2022-09-28:
> Hello,
> 
> I use Holybro Kakute H7 mini uploading kakute H7. It works anyway. But there is difference from H7 mini with H7. H7 has SD card, So, I could save the parameters.
> 
> Is there way to save parameters in flash memory for kakute h7 mini?
> 
> Thanks,

Hi, I just wonder which boot loader do you upload into the Kakute H7 mini. I use [holybro_kakuteh7_bootloader.hex](https://raw.githubusercontent.com/PX4/PX4-user_guide/main/assets/flight_controller/kakuteh7/holybro_kakuteh7_bootloader.hex) and successfully uploaded to the FCU. However, when I try to upload the firmware, it shows:
[image]

I have tried the same process with Kakute H7 and everything goes right, but for H7 mini, it does not work.

Any idea? Thanks in advance.

kokoory, 2022-09-28:
> Hi, I just wonder which boot loader do you upload into the Kakute H7 mini. I use [...]

I just follow the guide. I try to test it again.