PX4/PX4-Autopilot #20038: Where put macro to enable RS485?

FriedrichTuttas, 2022-08-11:
I want my Pixhawk 4 to communicate with servomotors over an RS485 interface. For that, I have a [communication module](https://digilent.com/reference/pmod/pmodrs485/reference-manual), which takes UART as an input and translates it to RS485. The Pixhawk needs to tell the module via the ~RE and DE pins whether it should read or write messages. I want to use the TELEM2 port of the Pixhawk to connect to the module and have its RTS pin connect to both ~RE and DE of the module.

Reading through the code of the file [stm32_serial.c](https://github.com/PX4/NuttX/blob/91bece51afbe7da9db12e3695cdbb4f4bba4bc83/arch/arm/src/stm32f7/stm32_serial.c) (lines 93, 187 and 965), I see that I need to set/alter a few macros for this to work:

- “CONFIG_USART3_RS485” needs to be defined.
- RXDMA/TXDMA and RS485 cannot be active on the same port at the same time, so “CONFIG_USART3_RXDMA” and “CONFIG_USART3_TXDMA” must not be defined.
- The direction pin that will connect to ~RE and DE must be set with the appropriate configuration through the macro “GPIO_USART3_RS485_DIR”.
- The direction polarity should be set to 1 via the macro “CONFIG_USART1_RS485_DIR_POLARITY”.

The macros that start with “CONFIG_” can handily be set in a defconfig file via the terminal command “make px4_fmu-v5 menuconfig”. However, the macro “GPIO_USART3_RS485_DIR” cannot be configured there. I found [this answer](https://github.com/PX4/PX4-Autopilot/issues/16187#issuecomment-913561070) on Github, which suggests to put the following line somewhere into the [board.h](https://github.com/PX4/PX4-Autopilot/blob/main/boards/px4/fmu-v5/nuttx-config/include/board.h) file:

#define GPIO_USART3_RS485_DIR (GPIO_OUTPUT | GPIO_PUSHPULL | GPIO_SPEED_50MHz | GPIO_PORTD | GPIO_PIN12 | GPIO_OUTPUT_CLEAR)

I put it in line [324](https://github.com/PX4/PX4-Autopilot/blob/3e688705474fe3047ddaee41946b6f42e8c938b8/boards/px4/fmu-v5/nuttx-config/include/board.h#L324) to group it with the other “#define GPIO_USART3…” macros.

Now the communication works as desired with the RTS pin of the TELEM2 port going HIGH before data is sent via the TX pin and going back to LOW right after.

My question is, whether this is the appropriate way to set up RS485 on my Pixhawk envisioned by whoever implemented those macros. Should the macro really be handwritten into the board.h file or is there some config file and terminal command to set it like for the other macros?

klxing, 2022-10-28:
Hi Friedrich，
Does your code work well to read and write through rs485?
I also encountered the problem to commonicate with rs485 servo,but when I configured the marcos as you described, the code return -1 when the function of write() iscalled.
here is how I write my code:
So can you please show me some example of how to read and write data in your application software?
Thanks a lot in advance!

FriedrichTuttas, 2022-10-31:
RS485 communication works fine for me. After setting the macros described in my original post, I followed [this guide](https://blog.mbedded.ninja/programming/operating-systems/linux/linux-serial-ports-using-c-cpp/) on how to communicate via UART.
You wrote 

> here is how I write my code:

but forgot to show any code. So I cannot help you there.

My code is scattered over multiple files and functions but I will try to dig up all the relevant parts, so you can see which implementation should work:
<details>
  <summary>Code for opening serial port</summary>

  I switched to a pixhawk 6X and use its TELEM3 port here. You will need to replace the string `"/dev/ttyS1"` with the appropriate device of the port you want to use.
  
  ```cpp
// Open Serial Port TELEM3
int serial_port = open("/dev/ttyS1", O_RDWR  | O_NOCTTY | O_NONBLOCK);
// Check for errors
if (serial_port < 0) {
	PX4_ERR("Error %i from open: %s\n", errno, strerror(errno));
	return nullptr;
}
struct termios tty;
if(tcgetattr(serial_port, &tty) != 0) {
	PX4_ERR("Error %i from tcgetattr: %s\n", errno, strerror(errno));
	return nullptr;
}
tty.c_cflag &= ~PARENB; // Clear parity bit, disabling parity
tty.c_cflag |= CSTOPB;  // Set stop field, two stop bits used in communication
tty.c_cflag &= ~CSIZE; // Clear all the size bits
tty.c_cflag |= CS8; // 8 bits per byte
tty.c_cflag &= ~CRTSCTS; // Disable RTS/CTS hardware flow control
tty.c_cflag |= CREAD | CLOCAL; // Turn on READ & ignore ctrl lines (CLOCAL = 1)
tty.c_lflag &= ~ICANON; // Diable canonical mode
tty.c_lflag &= ~ECHO; // Disable echo
tty.c_lflag &= ~ECHOE; // Disable erasure
tty.c_lflag &= ~ECHONL; // Disable new-line echo
tty.c_lflag &= ~ISIG; // Disable interpretation of INTR, QUIT and SUSP
tty.c_iflag &= ~(IXON | IXOFF | IXANY); // Turn off s/w flow ctrl
tty.c_iflag &= ~(IGNBRK|BRKINT|PARMRK|ISTRIP|INLCR|IGNCR|ICRNL); // Disable any special handling of received bytes
tty.c_oflag &= ~OPOST; // Prevent special interpretation of output bytes (e.g. newline chars)
tty.c_oflag &= ~ONLCR; // Prevent conversion of newline to carriage return/line feed
tty.c_cc[VTIME] = 10;  // Wait for up to 1s (10 deciseconds), returning as soon as any data is received.
tty.c_cc[VMIN] = 0;
// Set baud rate to 115200
cfsetispeed(&tty, B115200);
cfsetospeed(&tty, B115200);
// Save tty settings, also checking for error
if (tcsetattr(serial_port, TCSANOW, &tty) != 0) {
	PX4_ERR("Error %i from tcsetattr: %s\n", errno, strerror(errno));
	return nullptr;
}
  ```
  
</details>
<details>
  <summary>Code for writing messages</summary>

  ```cpp
unsigned char msg[6] = {1, 2, 3, 4, 5, 6}; // some random message
int num_bytes = write(serial_port, &msg, 6);
  ```
</details>
<details>
  <summary>Code for reading messages</summary>

  ```cpp
unsigned char read_buf[6];
int num_bytes = read(serial_port, &read_buf, 6);
```
</details>

klxing, 2022-10-31:
> 

Hi Friedrich, Thanks so much for your reply.
As my code went wrong I tried again as you posted above. However it still remains to return -1 when I call write() and cannot write or read data.
Here is my operation ,would you please check it for me when you are avaliable?
1.by running  "make px4_fmu-v2_default menuconfig" :
    uncheck "System Type -> U[S]ART Configuration -> USART3 Rx DMA"
    check "System Type -> U[S]ART Configuration -> RS-485 on USART3" and save the configuration
2.Added "#define GPIO_USART3_RS485_DIR (GPIO_OUTPUT | GPIO_PUSHPULL | GPIO_SPEED_50MHz | GPIO_PORTD | GPIO_PIN12 | GPIO_OUTPUT_CLEAR)" in board.h
3. Added the **Code for opening serial port**  you provided above. Since I am using TELEM on PIXHAWK 2.4.8 ,I use "/dev/ttyS2"instead,and the handler ID  got from open() is 3.
4. using read() and write()function to RX/TX date. Here the write() function returned -1 and I just cannot figure out why this happen.

here is my code when doing read and write:

int rs485_check(int uart_read)
{
    //char senddata[3] = {0xe8,0x02,0xbc};
    char senddata[3] = {0,0,1};
    unsigned char read_buffer[2];
    double dist;
    int tx_ret = 0;
    PX4_INFO("fd_uart_read: %d\n",uart_read);// here it prints fd_uart_read:3

    //tx_ret = write(uart_read,0,1);
    tx_ret = write(uart_read,&senddata,3); //both this scentence and the one above will return -1
    PX4_INFO("write_ret: %d\n",tx_ret);

    if(tx_ret >= 0)
        {
            printf("write first successful\n");
		    usleep(50);
            if(read(uart_read,&read_buffer,2) > 0)
		    {
            printf("read first successful\n");
            PX4_INFO("Raw Data[0]: %x\n",read_buffer[0]);
            PX4_INFO("Raw Data[1]: %x\n",read_buffer[1]);
 	  dist = double(read_buffer[1]<<8|read_buffer[0]);
            PX4_INFO("%f \n",dist);
            }
            else
            {
                PX4_INFO("read failed!");
            }
        }
        else
       {
            PX4_INFO("Write Failed!");
        }
	usleep(5000);
	return 0;
}

FriedrichTuttas, 2022-10-31:
If I understand your code correctly, you called the file descriptor `uart_read`. This suggests that you opened the uart device in read mode. In order to write to the device, you would need to open it in write (or read-write) mode. You can see in my code snippet that I used `O_RDWR` in the `open` function.

klxing, 2022-10-31:
> If I understand your code correctly, you called the file descriptor `uart_read`. This suggests that you opened the uart device in read mode. In order to write to the device, you would need to open [...]

actually i used the same way as yours to open it:
 int serial_fd = open(uart_name, O_RDWR  | O_NOCTTY | O_NONBLOCK);

I was wondering whether I left some definition or made some confliction, but just cannot find it,awful……

FriedrichTuttas, 2022-10-31:
If you named the file descriptor `serial_fd`, then the write command should say `write(serial_fd,&senddata,3)`.

klxing, 2022-10-31:
> 

it's **serial_fd** when opening and transfered to **uart_read**  as a parameter when the function **rs485_check()** is called.

 Now I am thinking am I setting wrong parameter "GPIO_PORTD | GPIO_PIN12" ？ what does this pin mean? In pixhawk 2.4.8 this pin relates to USART3-RTS. which pin should I chose?

FriedrichTuttas, 2022-10-31:
It should be the RTS pin. But this should not be relevant for the UART part of the communication. I do not think this can cause the error that you receive.

You can have a look at the error string with something like this:
```cpp
PX4_ERR("Error writing: %s", strerror(errno));
```

Put it right after the `write` command!

klxing, 2022-10-31:
> It should be the RTS pin. But this should not be relevant for the UART part of the communication. I do not think this can cause the error that you receive.
> 
> You can have a look at the error string with something like this:
> 
> ```c++
> PX4_ERR("Error writing: %s", strerror(errno));
> ```
> 
> Put it right after the `write` command!

Drat! It says "ERROR [ult_485] Error writing: Unknown error" when I do this ……

FriedrichTuttas, 2022-10-31:
You do include these three files?
```cpp
#include <unistd.h> // write(), read(), close()
#include <termios.h>
#include <px4_platform_common/log.h>
```

klxing, 2022-10-31:
the first two are included. And it makes no difference when I add the third one ……

FriedrichTuttas, 2022-10-31:
Then, I do not know.

klxing, 2022-10-31:
> Then, I do not know.

Well, Thank you very much all the same for so much help!

Or, could you send me an example of your module using rs485 so I can compare and have a test? If it won't violate your privacy or technical secret.

klxing, 2022-11-02:
> Then, I do not know.

Hey Friedrich, I am studying the code these days, as I can see from your code, you disabled software flow control by setting :tty.c_iflag &= ~(IXON | IXOFF | IXANY);
My question is , do you need to wire the RTS pin with the DE/RE pins on your convert module? or just leave them unconeected? 
Can this be the problem which caused my write failure.

klxing, 2022-11-03:
After all these suffering in trying to read rs485 data, it turns out that it's just because I made a really stupid mistake.
I put open() and init() function of rs485 in the Module::init() and the read() and write() function in Module::Run()
This causes the mistake when calling read() and write(). Now when I put open(), init() , read() and write() all in Module::Run(), it worked as desired.
Thanks for your kind help! 
There may be nobody made such mistake as I did but I want to publish it and draw this issue to a conclusion.

Also, I tested that using some ttl to 485 converters that don't require the signal of RTS control, the pin can be left blank, it works  well for me.

FriedrichTuttas, 2022-11-03:
> My question is , do you need to wire the RTS pin with the DE/RE pins on your convert module? 

Yes, if you want to be able to do both: send and receive. Otherwise, you can connect the ~RE and DE pins constantly to ground or voltage.

> some ttl to 485 converters that don't require the signal of RTS control, the pin can be left blank

I am no expert but I do not think it is healthy for the converter and the connected devices.