PX4/PX4-Autopilot #20039: How to auto-start work queue task applications?

FriedrichTuttas, 2022-08-11:
I want to write a module in PX4 that runs right from the start of the Pixhawk. I found [this page](https://docs.px4.io/main/en/modules/module_template.html) of the PX4 documentation, which describes work tasks and work queue tasks. For work tasks, it states that their call (`taskname start`) must be added to the system’s startup script. For work queue tasks, it says:

> A work queue task application is just the same as an ordinary (task) application, except that it needs to specify that it is a work queue task, and **schedule itself to run during initialisation**.

To me “schedule itself to run during initialisation” sounds exactly like what I want to achieve. However, the page does not mention how this is accomplished. Even the example module from the page does not run from the start of the Pixhawk. I have to manually execute `work_item_example start` in the MAVLink console of QGroundcontrol for it to run.

So, my question is: How can one create a module as a work queue task that runs from the start of the Pixhawk?

Ncerzzk, 2022-08-17:
you can directly add the `XXXX start` into rcS script(which is the startup script)

FriedrichTuttas, 2022-08-17:
Thanks for the answer. So, a work queue task is no different from a normal task when it comes to automatic start? Then, why does the page I cited say that a key difference between the two is that a work queue task must "schedule itself to run during initialisation"?

Ncerzzk, 2022-08-17:
> So, a work queue task is no different from a normal task when it comes to automatic start? 

Yes, I think so.

> Then, why does the page I cited say that a key difference between the two is that a work queue task must "schedule itself to run during initialisation"?

because of the different implement theory, a normal task behave like a thread, and it could be scheduled by os(nuttx, linux...), 
while work queue task could not, and it must be added to a queue(according to priority) and be handled by a worker thread.

closed 2025-10-27 by github-actions[bot]