PX4/PX4-Autopilot #20041: Cannot catkin_make on Ubuntu 18.04 with Python 2.7

AlexWUrobot, 2022-08-12:
## Describe the bug
Follow the website to build: MAVROS Offboard control example (Python) (ROS1)
https://docs.px4.io/main/en/ros/mavros_offboard_python.html

roslaunch offboard_py start_offb.launch works
roslaunch px4 mavros_posix_sitl.launch works

The gazebo simulation can operate well and print all mavros rostopics, but when rostopic echo these topic,
there is no data publishing.
Thus, suspect the reason may be due to the catkin_make the PX4-Autopilot failed on Ubuntu 18.04 with Python 2.7
**Try to catkin make the PX4-Autopilot on Ubuntu 18.04 with Python 2.7, but fail**

## To Reproduce
Steps to reproduce the behavior:
1. git clone the repository 
2. catkin_make
3. already try to solve many issues like these
https://github.com/cjhutto/vaderSentiment/issues/47

## Expected behavior
can control the drone to take off in the gazebo simulation

## Log Files and Screenshots
[image]

## Drone (please complete the following information):
- Pixhawk 4 (FMUv5 )

Thank you very much for the your help.  really appreciate it

Jaeyoung-Lim, 2022-08-12:
@AlexWUrobot Is there a specific reason you are using catkin_make? It should work with `catkin build`

AlexWUrobot, 2022-08-12:
@Jaeyoung-Lim ,thank you very much for the reply.  
I also tried to catkin build and get similar error, 
[image]

so I guess that I need to catkin build on python3, instead of python2, which is default on Ubuntu 18.04 (melodic) 
Could I inquire whether the document is this https://github.com/PX4/PX4-Autopilot/tree/main/Documentation?

Jaeyoung-Lim, 2022-08-12:
Yes, you should take px4 out of the catkin workspace

AlexWUrobot, 2022-08-12:
Thanks for the reply :  )
I tried to read the document but still cannot find how to catkin build in this document https://github.com/PX4/PX4-Autopilot/tree/main/Documentation?
[image]

I git clone the PX4-Autopilot in my workspace ($~/PX4_ws/src/PX4-Autopilot/.....)
[image]
What do you mean that take px4 out of the catkin workspace?

AlexWUrobot, 2022-08-12:
If take px4 (PX4-Autopilot) out of the catkin workspace, it looks wrong, because PX4-Autopilot is not a workspace. 
 I think PX4-Autopilot should be a package, because it has its own cmake list and launch
[image]

Jaeyoung-Lim, 2022-08-12:
@AlexWUrobot You can follow the instructions here: https://docs.px4.io/main/en/simulation/ros_interface.html#launching-gazebo-with-ros-wrappers

You should not build the package with catkin since catkin requires a project to have certain directory structures. The instructions above lets you build px4 independently and then get it discoverable in the package path

AlexWUrobot, 2022-08-15:
Thanks for the information. really appreciate 
I followed the instructions and completed every step without error.
[image]
Next, tried to 
roslaunch roslaunch px4 mavros_posix_sitl.launch
There is no error and rostopic list can show many topics.
However, when rostopic echo to any specific topic, there is no data.
[image]

P.S. I checked the px4 and bashrc.
The PX4-Autopilot has been independently built by the "DONT_RUN=1 make px4_sitl_default gazebo
" from the instruction: https://docs.px4.io/main/en/simulation/ros_interface.html#launching-gazebo-with-ros-wrappers
[image]

AlexWUrobot, 2022-08-15:
[image]
[image]