PX4/PX4-Autopilot #20042: symforce>=0.5.0

surepay123456, 2022-08-12:
When bash ./PX4-Autopilot/Tools/setup/ubuntu.sh
this happened:
Could not find a version that satisfies the requirement symforce>=0.5.0 (from -r /home/ubuntu18.04/PX4-Autopilot/Tools/setup/requirements.txt (line 29)) (from versions: )
No matching distribution found for symforce>=0.5.0 (from -r /home/ubuntu18.04/PX4-Autopilot/Tools/setup/requirements.txt (line 29))

JulienRineau, 2022-08-13:
I have the same problem. Did you solved it ?

surepay123456, 2022-08-15:
> I have the same problem. Did you solved it @surepay123456 
Hi Julien. I git the stable branch  release/1.11 of PX4. It did't require symforce>=0.5.0 , and it worked.

bresch, 2022-08-15:
@surepay123456  , @JulienRineau , we'll set this requirement as optional as it isn't required to build the code, it's just to (re)generate some equations for the estimators (that are already generated in the repo)

closed 2022-08-16 by dagar