PX4/PX4-Autopilot #20045: Increase GPS input update rate to 20 HZ

moreba1, 2022-08-13:
Hi

I think default input rate of gps sensor is 5 hz constant. Is there possible that we be able to change input rate ?

https://github.com/PX4/PX4-GPSDrivers/blob/main/src/ubx.cpp#L464

thanks

bkueng, 2022-08-15:
It depends on the GPS, the F9 select 8-10Hz. They can go to 20Hz when disabling some satellite systems.
The estimator will interpolate and gives you a much higher rate already. Isn't that enough?

moreba1, 2022-08-15:
> It depends on the GPS, the F9 select 8-10Hz. They can go to 20Hz when disabling some satellite systems. The estimator will interpolate and gives you a much higher rate already. Isn't that enough?

Does autopilot select gps input rate or Update rate set by gps and autopilot only give it?

bkueng, 2022-08-16:
Not sure I understand the question, the gps driver configures the GPS device and the estimator then uses the published data.

moreba1, 2022-08-17:
> Not sure I understand the question, the gps driver configures the GPS device and the estimator then uses the published data.

Which rate gps driver set if i connect a f9p?
8hz or 10hz or 20 hz

bkueng, 2022-08-17:
10hz

moreba1, 2022-08-17:
> 10hz

Ok, is there possible that i change It to 20 hz? 
By change driver code