PX4/PX4-Autopilot #20046: Feature TODO list for full Ignition Gazebo Support

Jaeyoung-Lim, 2022-08-14:
## Describe problem solved by the proposed feature
The following list outlines what needs to be ported over to support ignition gazebo

### Utilities
- [ ] Camera manager plugin

### Models
- [ ] Gimbal Controller and gimbal models
- [ ] Parachute plugin
- [ ] Lift drag plugin and

### Sensors
- [ ] Airspeed sensors
- [] Distance Sensor

SangwooJ, 2023-06-02:
@Jaeyoung-Lim  I am trying to add lidar(gpu_ray) to the x500 model in Gazebo(ignition) by following this instructions https://gazebosim.org/docs/garden/sensors . However, It is not working properly, the simulator stops and crashes. Is this a PX4 problem?

Jaeyoung-Lim, 2023-06-02:
@SangwooJ I don't think so

SangwooJ, 2023-06-05:
@Jaeyoung-Lim Thanks for the reply. Currently is it possible to add Lidar(gpu_ray) to the x500 in gazebo ignition(garden)?

closed 2024-03-11 by Jaeyoung-Lim