PX4/PX4-Autopilot #20047: problem when conducting "make px4_sitl gazebo"

ROBERTJEFFLV, 2022-08-14:
## Hi, sorry I met a problem when conducting "make px4_sitl gazebo". After following the guide till i typed 'make px4_sitl gazebo 'The output shows as below. (Literally tons of error popped out...the log copied is just small part of them)

  143 |   PROTOBUF_NDEBUG_INLINE T* mutable_unknown_fields() {
      |                             ^~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/google/protobuf/metadata_lite.h:143:29: note:   template argument deduction/substitution failed:
In file included from /usr/include/gazebo-9/gazebo/msgs/MessageTypes.hh:128,
                 from /usr/include/gazebo-9/gazebo/rendering/Visual.hh:38,
                 from /usr/include/gazebo-9/gazebo/rendering/ApplyWrenchVisual.hh:23,
                 from /usr/include/gazebo-9/gazebo/rendering/rendering.hh:2,
                 from /home/robert_lv_/PX4-Autopilot/Tools/sitl_gazebo/include/force_visual/force_visual.h:5,
                 from /home/robert_lv_/PX4-Autopilot/Tools/sitl_gazebo/src/force_visual/force_visual.cpp:1:
/usr/include/gazebo-9/gazebo/msgs/wind.pb.h:100:55: note:   couldn’t deduce template parameter ‘T’
  100 |     return _internal_metadata_.mutable_unknown_fields();
      |                                                       ^
In file included from /usr/include/gazebo-9/gazebo/msgs/MessageTypes.hh:129,
                 from /usr/include/gazebo-9/gazebo/rendering/Visual.hh:38,
                 from /usr/include/gazebo-9/gazebo/rendering/ApplyWrenchVisual.hh:23,
                 from /usr/include/gazebo-9/gazebo/rendering/rendering.hh:2,
                 from /home/robert_lv_/PX4-Autopilot/Tools/sitl_gazebo/include/force_visual/force_visual.h:5,
                 from /home/robert_lv_/PX4-Autopilot/Tools/sitl_gazebo/src/force_visual/force_visual.cpp:1:
/usr/include/gazebo-9/gazebo/msgs/wireless_node.pb.h: At global scope:
/usr/include/gazebo-9/gazebo/msgs/wireless_node.pb.h:46:46: error: ‘FieldMetadata’ in namespace ‘google::protobuf::internal’ does not name a type; did you mean ‘AnyMetadata’?
   46 |   static const ::google::protobuf::internal::FieldMetadata field_metadata[];
      |                                              ^~~~~~~~~~~~~
      |                                              AnyMetadata
/usr/include/gazebo-9/gazebo/msgs/wireless_node.pb.h:47:46: error: ‘SerializationTable’ in namespace ‘google::protobuf::internal’ does not name a type
   47 |   static const ::google::protobuf::internal::SerializationTable serialization_table[];
      |                                              ^~~~~~~~~~~~~~~~~~
In file included from /usr/include/gazebo-9/gazebo/msgs/MessageTypes.hh:129,
                 from /usr/include/gazebo-9/gazebo/rendering/Visual.hh:38,
                 from /usr/include/gazebo-9/gazebo/rendering/ApplyWrenchVisual.hh:23,
                 from /usr/include/gazebo-9/gazebo/rendering/rendering.hh:2,
                 from /home/robert_lv_/PX4-Autopilot/Tools/sitl_gazebo/include/force_visual/force_visual.h:5,
                 from /home/robert_lv_/PX4-Autopilot/Tools/sitl_gazebo/src/force_visual/force_visual.cpp:1:
/usr/include/gazebo-9/gazebo/msgs/wireless_node.pb.h:204:33: error: ‘InternalMetadataWithArena’ in namespace ‘google::protobuf::internal’ does not name a type; did you mean ‘InternalMetadata’?
  204 |   ::google::protobuf::internal::InternalMetadataWithArena _internal_metadata_;
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                                 InternalMetadata
/usr/include/gazebo-9/gazebo/msgs/wireless_node.pb.h:120:24: error: ‘gazebo::msgs::WirelessNode* gazebo::msgs::WirelessNode::New() const’ marked ‘final’, but is not virtual
  120 |   inline WirelessNode* New() const final {
      |                        ^~~
/usr/include/gazebo-9/gazebo/msgs/wireless_node.pb.h:135:8: error: ‘bool gazebo::msgs::WirelessNode::MergePartialFromCodedStream(google::protobuf::io::CodedInputStream*)’ marked ‘final’, but is not virtual
  135 |   bool MergePartialFromCodedStream(
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-9/gazebo/msgs/wireless_node.pb.h:137:8: error: ‘void gazebo::msgs::WirelessNode::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const’ marked ‘final’, but is not virtual
  137 |   void SerializeWithCachedSizes(
      |        ^~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-9/gazebo/msgs/wireless_node.pb.h:139:30: error: ‘google::protobuf::uint8* gazebo::msgs::WirelessNode::InternalSerializeWithCachedSizesToArray(bool, google::protobuf::uint8*) const’ marked ‘final’, but is not virtual
  139 |   ::google::protobuf::uint8* InternalSerializeWithCachedSizesToArray(
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/gazebo-9/gazebo/msgs/wireless_node.pb.h: In member function ‘const google::protobuf::UnknownFieldSet& gazebo::msgs::WirelessNode::unknown_fields() const’:
/usr/include/gazebo-9/gazebo/msgs/wireless_node.pb.h:96:47: error: no matching function for call to ‘google::protobuf::internal::InternalMetadata::unknown_fields() const’
   96 |     return _internal_metadata_.unknown_fields();
      |                                               ^
In file included from /usr/local/include/google/protobuf/message_lite.h:54,
                 from /usr/local/include/google/protobuf/generated_enum_util.h:38,
                 from /usr/local/include/google/protobuf/map.h:60,
                 from /usr/include/google/protobuf/generated_message_table_driven.h:34,
                 from /usr/include/ignition/msgs1/ignition/msgs/altimeter.pb.h:25,
                 from /usr/include/ignition/msgs1/ignition/msgs/MessageTypes.hh:31,
                 from /usr/include/gazebo-9/gazebo/rendering/Visual.hh:32,
                 from /usr/include/gazebo-9/gazebo/rendering/ApplyWrenchVisual.hh:23,
                 from /usr/include/gazebo-9/gazebo/rendering/rendering.hh:2,
                 from /home/robert_lv_/PX4-Autopilot/Tools/sitl_gazebo/include/force_visual/force_visual.h:5,
                 from /home/robert_lv_/PX4-Autopilot/Tools/sitl_gazebo/src/force_visual/force_visual.cpp:1:
/usr/local/include/google/protobuf/metadata_lite.h:133:35: note: candidate: ‘template<class T> const T& google::protobuf::internal::InternalMetadata::unknown_fields(const T& (*)()) const’
  133 |   PROTOBUF_NDEBUG_INLINE const T& unknown_fields(
      |                                   ^~~~~~~~~~~~~~
/usr/local/include/google/protobuf/metadata_lite.h:133:35: note:   template argument deduction/substitution failed:
In file included from /usr/include/gazebo-9/gazebo/msgs/MessageTypes.hh:129,
[... 423 lines cut ...]
/usr/include/gazebo-9/gazebo/msgs/wrench_stamped.pb.h:101:55: note:   couldn’t deduce template parameter ‘T’
  101 |     return _internal_metadata_.mutable_unknown_fields();
      |                                                       ^
In file included from /usr/include/ignition/msgs1/ignition/msgs.hh:24,
                 from /usr/include/ignition/transport4/ignition/transport/Node.hh:32,
                 from /usr/include/gazebo-9/gazebo/rendering/Projector.hh:26,
                 from /usr/include/gazebo-9/gazebo/rendering/rendering.hh:34,
                 from /home/robert_lv_/PX4-Autopilot/Tools/sitl_gazebo/include/force_visual/force_visual.h:5,
                 from /home/robert_lv_/PX4-Autopilot/Tools/sitl_gazebo/src/force_visual/force_visual.cpp:1:
/usr/include/ignition/msgs1/ignition/msgs/Generator.hh: At global scope:
/usr/include/ignition/msgs1/ignition/msgs/Generator.hh:46:21: error: ‘string’ does not name a type; did you mean ‘strings’?
   46 |               const string &_parameter,
      |                     ^~~~~~
      |                     strings
/usr/include/ignition/msgs1/ignition/msgs/Generator.hh:48:15: error: ‘string’ has not been declared
   48 |               string *_error) const;
      |               ^~~~~~
ninja: build stopped: subcommand failed.
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /home/robert_lv_/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build /home/robert_lv_/PX4-Autopilot/build/px4_sitl_default/build_gazebo -- -j 8
ninja: build stopped: subcommand failed.
make: *** [Makefile:235：px4_sitl] error 1

I'm new to linux and literally no CS relevant experiences... So... **PLEASE HELP!!    T ^ T**
Thanks!

junwoo091400, 2022-12-02:
@ROBERTJEFFLV could you resolve it?

ROBERTJEFFLV, 2022-12-02:
nope.

julianoes, 2023-05-18:
Closing because this thread is outdated and stale.

@Abner0907 please create a new issue, post the steps you did to install the dependencies (e.g. https://docs.px4.io/main/en/dev_setup/dev_env_linux_ubuntu.html) and what exactly the error is with what command.

closed 2023-05-18 by julianoes