PX4/PX4-Autopilot #20051: wrong unit in MS5837.cpp

FlorianPix, 2022-08-15:
## Describe the bug:

in line 335 of MS5837.cpp:

`/* pressure calculation, result in Pa */`

`int32_t P = (((raw * _SENS) >> 21) - _OFF) >> 13;`

this is different from [what the sensor manufacturer says](https://www.te.com/commerce/DocumentDelivery/DDEController?Action=showdoc&DocId=Data+Sheet%7FMS5837-30BA%7FB1%7Fpdf%7FEnglish%7FENG_DS_MS5837-30BA_B1.pdf%7FCAT-BLPS0017):
[image]

the `/10` was left out at the end so instead of mBar it would be cBar = 10⁻² Bar = 10⁻² * 10³ hPa = 10 hPa

This means that the pressure result is off by a factor of 10.

## Solution:

add `*10` to get hPa (which is the unit used by EKF2 and would be more convenient
or `/ 10` to get Pa as suggested in the comment
I personally would prefer the first option.

FlorianPix, 2022-08-22:
@dagar Actually I had to add *10 at the end of that line to get the right result. I couldn't find any other deviations in your code from the flow chart tough so im not sure if the flowchart is just wrong. 

Changed it to:
`int32_t P = ((((raw * _SENS) >> 21) - _OFF) >> 13) * 10;`

VehicleAirData.cpp: PressureToAltitude: 
```
// liquid pressure to altitude

// pressure at sea level (depends on the weather: extreme min = 870 hPa, average: 1013.25, extreme max = 1084.8 hPa)
// can be set by changing SENS_BARO_QNH px4 param

// * 100 to get from hPa to Pa
const float p1 = _param_sens_baro_qnh.get() * 100;

// pressure_measured in Pa
const float p0 = pressure_pa;

// rho source https://bluerobotics.com/learn/pressure-depth-calculator/#hydrostatic-water-pressure-formula
// we don't adjust rho for temperature, salinity or air content
// except for different constant values for fresh/salt water
float rho;
if (_param_baro_medium.get() == 1){
    // rho in fresh water in kg/m3
    rho = 997.0474;
} else {
    // rho in salt water in kg/m3
    rho = 1023.6;
}

// altitude = (pressure_measured - pressure_sea_level) / (liquid_density * gravity);
// but we want depth so we switch the pressures:
// depth = (pressure_sea_level - pressure_measured) / (liquid_density * gravity);
// has the added bonus that it complies to the NED or FRD reference frame

// g/gravity i.e. CONSTANTS_ONE_G = 9.80665 m/s2
// we don't adjust g for altitude or latitude at the moment

altitude = (p0 - p1) / (rho * CONSTANTS_ONE_G);
```

spiderkeys, 2022-08-22:
Can confirm and have been operating with this change for months. Apparently we had made a similar change 4-5 months ago:
```c++
	} else {
		// Pressure calculation, result in (Pa/10)
		int32_t P = (((raw * _SENS) >> 21) - _OFF) >> 13;

		// Convert to pascals
		float pressure = P * 10.0f;

		_px4_fluid_pressure.update(timestamp_sample, pressure);
	}
```

closed 2025-11-13 by MaEtUgR