PX4/PX4-Autopilot #20073: use different height/depth calculation for uuv's

FlorianPix, 2022-08-19:
## Describe problem solved by the proposed feature

I am using the MS5837 sensor for my uuv which gives me a pressure reading that I want to use to estimate the depth in water.
EKF2 uses the result from VehicleAirData::PressureToAltitude but that one uses an equation that is only correct if the vehicle is in air.
Under water a different equation needs to be used.

## Describe your preferred solution

Add a param `IN_AIR` to VehicleAirData or use an existing one (maybe vehicle type).
If IN_AIR is false use this equation: `altitude = (measure_p - p_at_sea_level) * 0.01f;`

bresch, 2022-08-22:
We could even have a baro parameter to select between:
- Air
- Salt water
- Fresh water

(https://bluerobotics.com/learn/pressure-depth-calculator/#hydrostatic-water-pressure-formula)

FlorianPix, 2022-08-22:
I think that would be enough although it would be nice to measure the actual density of the surrounding water. The density of water depends on temperature, salinity and air content but that's a [rabbit hole](https://link.springer.com/content/pdf/bbm%3A978-3-319-18908-6%2F1.pdf) we probably don't want to go down just for some small accuracy improvements.
Maybe we could use the standard salt water/ fresh water density as a first guess and increase or lower it a bit with the temperature. 

ρ(T) = ρ₀ + a₁*T + a₂*T² + a₃*T³ + a₄*T⁴ + a₅*T⁵,

where the temperature T is in °C, and the values of coefficients are the following:

    ρ₀ = fresh/salt water pressure estimate
    a₁ = 0.0752 kg/(m³·°C),
    a₂ = -0.0089 kg/(m³·°C²),
    a₃ = 7.36413*10⁻⁵ kg/(m³·°C³),
    a₄ = 4.74639*10⁻⁷ kg/(m³·°C⁴),
    a₅ = 1.34888*10⁻⁹ kg/(m³·°C⁵),

[source](https://www.omnicalculator.com/physics/water-density)

spiderkeys, 2022-08-22:
Just wanted to provide another perspective on this issue. Our team went a slightly different route and added support for this in the following way:
- Introduced a new sensor type PX4FluidPressure
- Made the MS5837 driver implement that sensor type
- Introduced VehicleHydroData as a parallel concept to VehicleAirData, which uses PX4FluidPressure sensors as one (of potentially many) inputs for calculating different types of altitudes
- Modified EKF2 to support the use of VehicleHydroData alongside VehicleAirData

One of the reasons we went this route is to allow both VehicleAirData and HydroData to coexist simultaneously while allowing an external discriminator to make a decision about whether or not the vehicle is "in water/a fluid" and switch which paradigm is being applied in EKF2. This would help support vehicle types that can transition between air and water environments, such as amphibious vehicles.

Another reason is one you touched on, which is that according to the UNESCO formulas, there are many more inputs relating to calculating depth than just a fluid pressure input and specific gravity value. For ROVs, these values are usually not that important, but for AUVs or missions that are specifically targeting the collection of scientific data in the water column, they can become important, so we wanted to make it possible to support this. 

Below is a rough diagram of how we mapped out similar/equivalent concepts in PX4 between air and under-water/liquid concerns and the common types of sensors that provide related inputs for underwater use-cases:

[image]

None of this is to say that I feel our approach is better/more appropriate than adding various parameters/branches in the existing drivers/code; I just wanted to share our approach, as it has been working fairly well for us so far on a wide range of underwater vehicle sizes and types.

spiderkeys, 2022-08-22:
One other thing that is worth thinking about when overloading barometers with the task of calculating depth in water is that it is very handy to have internal barometers in UUVs. 

We use the MS5837 (and Keller LD4) for external fluid pressure, while using one or more internal barometers that support 20kPa to 110kPa ranges to support leak detection. It is fairly common to pull vacuums inside of the pressure vessels that house the electronics, and in this configuration, internal barometers can be used to detect slow leaks or manufacturing/assembly defects (compromised o-rings and such), or sudden sharp pressure increases during operation (assuming the exact nature of the failure allows time for detection + message transmission of the failure). There have definitely been a few cases where a seal failure could be detected via internal pressure long before water reached one of the leak-detection GPIO-based sensors that are commonly used.

FlorianPix, 2022-08-22:
Very intressting ! Could you join the next devcall (24.08.) to talk about this ?

spiderkeys, 2022-08-22:
Will do!

Hardwarize, 2026-08-19:
Hi, I'm also interested in this feature, can I do something on  my side to help?

farhangnaderi, 2026-08-19:
@Hardwarize  Our UUV needs some rework to be done. I and @Pedro-Roque will be happy review PRs relevant to it for the time being.

Hardwarize, 2026-08-24:
Thanks for your answer, @farhangnaderi.

After taking a look at the issue, I think the solution proposed by @spiderkeys is highly achievable. While it might require a bit of extra work to handle every edge case (for example the one about manage internal and external pressures simultaneously), establishing a dedicated structure for underwater navigation would be very useful for people working in this field or researching, myself included. At least as I see the issue It would make PX4 a much more attractive platform for marine use cases.

From my point of view, while the first approach, the one about reusing the params, is easier to implement and sufficient to get accurate EKF2 estimations for simple setups, I would like to give @spiderkeys' parallel approach a try since there is no immediate urgency. I know that PX4 is a serious project used in commercial products, and for professional UUVs operating at +100m depths, very coomon scenario in real world missions, laying the groundwork for UNESCO formulas will give us a significant advantage.

If you agree with this direction, I can start working on the VehicleHydroData structure as a first step. I know this approach involves more work on the EKF2 side eventually, but since I am not a PX4 expert yet, defining the uORB topic feels like a solid starting point.

My idea is to create a uORB topic, equivalent to VehicleAirData, that provides the following information:

* uint64 timestamp                    # time since system start (microseconds)
* uint64 timestamp_sample             # the timestamp of the raw data (microseconds)

* uint32 depth_sensor_device_id       # unique device ID for the external depth/pressure sensor

* float32 pressure_pa                 # absolute hydrostatic pressure (Pascals)
* float32 surface_pressure_pa         # atmospheric pressure reference at surface (Pascals)

* float32 depth_meter                 # depth below surface, positive downwards (meters)
* float32 alt_meter                   # altitude relative to surface in NED frame, negative downwards (meters)
* float32 depth_variance              # variance of depth measurement (m^2)

* float32 water_temperature           # ambient water temperature (degrees Celsius)
* uint8 temperature_source            # source: 0: default, 1: depth sensor, 2: external probe

* float32 salinity                    # practical salinity (PSU or ppt)
* float32 rho                         # computed water density (kg/m^3)
* float32 speed_of_sound_m_s      

* uint8 calibration_count             # increments whenever tare or calibration is updated

I'm not an expert on the UNESCO formulas, but after doing some research and discussing it with ChatGPT, I understand that the calculations are computationally heavy and could be difficult to perform at 250 Hz in the main EKF loop. However, we can figure out the calculation scheduling later (e.g., computing density in a slower background task). At the very least, establishing this uORB topic now sets up a clean architecture, and even if we don't populate or use every single field right away, we will be in a great position to take advantage of them in the future.

Do you think that is a good or at least valid approach?
Any comment will be more than welcome

Hardwarize, 2026-09-16:
Hi @Pedro-Roque,
Thanks for your message! I've been working on the proposal, you can review the full document [here](https://app.notion.com/p/Github-document-3dbc01e6b15980e69476cd2124339c4b?source=copy_link). As you can see I've simplified the info that VehicleHydroData carry, no problem in add it again, I just want to make the first PR, if I achieve this point, as simple as possible.

Regarding which state to estimate: for now, the focus is strictly on depth. Inside EKF2, this translates directly to the vehicle's vertical position ("height" state, representing depth underwater).

I thought about whether other states could be derived from water data, but there aren't any immediate candidates worth adding to this specific pipeline. For instance, DVL systems already output direct velocity vectors, so that data, If I'm not wrong,  can simply be routed through PX4's existing vehicle_odometry or external vision pipelines.

However, by establishing the vehicle_hydro_data pipeline now, we create a modular foundation. If we want to expand UUV-specific features in the future, extending this architecture will be relatively easy.

More than happy to hear any feedback for the document or the approach in general.