PX4/PX4-Autopilot #20081: use param for disarmed value of ESCs connected via CAN

FlorianPix, 2022-08-20:
## Describe problem solved by the proposed feature
in v1.13 `PX4-Autopilot/src/drivers/uavcan/actuators/esc.cpp` line 91 is responsible for setting the disarmed value that gets published to the ESCs over CAN. The value is hardcoded to 0 but some ESCs don't idle at 0. For example my ESCs spin backwards with full power at 0.

## Describe your preferred solution
Use a param like PWM_MAIN_DIS to configure the CAN ESCs disarmed value.

FlorianPix, 2022-08-21:
just adding `disarmed: { min: 0, max: 8191, default: 4096 }` above [these lines](https://github.com/PX4/PX4-Autopilot/blob/889a2fddea152f745090669396d279ef254df5dd/src/drivers/uavcan/module.yaml#L15-L17) isn't enough. That adds the parameters but they aren't used then to actually change the value used by uavcan.

FlorianPix, 2022-08-21:
https://github.com/PX4/PX4-Autopilot/blob/main/src/drivers/uavcan/uavcan_main.cpp#L577
because the param didn't exist before it was set in uavcan_main.cpp and thereby invalidated afterwards

FlorianPix, 2022-08-21:
I still need to investigate why uavcan only reacts to actuator_motors and actuator_servo commands between 0 and 1 (leading to output 4096 and 8191). Because I need it to react to -1 till 0 (0 till 4096) as well.

FlorianPix, 2022-08-21:
As stated in [dronecan](https://github.com/dronecan/DSDL/blob/6bf9a446e553985a4eeaa7c1fb478a0a2253d8a1/uavcan/equipment/esc/1030.RawCommand.uavcan#L2) the usual range for CAN ESCs is -8191 to 8191. The ESCs im using might just not comply to the standard. One could still add the parameters with range -4096 to 4096 and set the default to 0.

bkueng, 2022-08-22:
Yes you need to remove this line as well if you want it to be confiurable: https://github.com/PX4/PX4-Autopilot/blob/main/src/drivers/uavcan/uavcan_main.cpp#L577
But I'd much rather have the ESC conform to the standard. Can you ask the manufacturer?

You can configure motors to be bidirectional in the UI: https://docs.px4.io/main/en/config/actuators.html

TorbjornHouge, 2023-01-03:
I would also prefer to have a UAVCAN_EC_DISx parameter like PWM_MAIN_DISx, and the option of outputting this upon pre-arm like the PWM-ports do. Currently, since my ESCs expect to be fed a zero-value (pwm 1000 or equal) they will not arm until I perform an ARM-DISARM-sequence. Upon the next ARM they will go to the idle-speed set in UAVCAN_EC_MINx. I can set this value to 1 in order to make them arm at the appropriate time, but that will not give me idle-speed when arming and it will then make the throttle response more jumpy in manual/stabilize.