PX4/PX4-Autopilot #20087: WARN  [PreFlightCheck] Mode not suitable for takeoff

bigbigpark, 2022-08-22:
## Describe the bug
I have installed and built `FastDDS` and `ROS2` on Ubuntu 20.04. <br/>

First terminal: `make px4_sitl_rtps gazebo`. <br/>
Second terminal: `micrortps_agent -t UDP`
When I type `ros2 topic list`, the running topic appears like the below screenshots. <br/>
[image]

Now I'm going to publish command to Quadrotor. <br/>
Trying to run `offboard control node` in my third terminal, it says:<br/>
~~~bash
Starting offboard control node...
[INFO] [1661166107.293669783] [offboard_control]: Arm command send
~~~
and in my first terminal,
~~~bash
WARN  [PreFlightCheck] Mode not suitable for takeoff
INFO  [commander] Failsafe mode deactivated	
INFO  [tone_alarm] notify negative
~~~

I'm pretty new for PX4 SITL, does anyone who can fix this? <br/>
Best regards.

Jaeyoung-Lim, 2022-08-22:
@bigbigpark What setpoints are you sending to use offboard mode? and have you tried taking off and then switching to offboard?

bigbigpark, 2022-08-23:
Thanks for replying. <br/>
I solved this problem #19919 as you mentioned.

closed 2022-08-23 by bigbigpark