PX4/PX4-Autopilot #20101: Unable to launch script

fatburg3r, 2022-08-24:
Hi, I'm a beginner and just started learning px4 and mavros. I wanted to launch a python script obtained from the px4 User Guide website. [https://docs.px4.io/main/en/ros/mavros_offboard_python.html](url)

I created a ros package (package_px4), python script file (position.py), launch file (start.launch). As per the website, I added 
```
source ~/Firmware/Tools/setup_gazebo.bash ~/Firmware ~/Firmware/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/Firmware
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/Firmware/Tools/sitl_gazebo
export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins
```
at the end of the .bashrc file. 

I tried to source the .bashrc file but the terminal gave me this output.
```
GAZEBO_PLUGIN_PATH :/usr/lib/x86_64-linux-gnu/gazebo-9/plugins:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins:/home/syuant/Firmware/build/px4_sitl_default/build_gazebo:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins:/home/syuant/Firmware/build/px4_sitl_default/build_gazebo:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins:/home/syuant/Firmware/build/px4_sitl_default/build_gazebo:/home/syuant/Firmware/build/px4_sitl_default/build_gazebo:/home/syuant/Firmware/build/px4_sitl_default/build_gazebo:/home/syuant/Firmware/build/px4_sitl_default/build_gazebo:/home/syuant/Firmware/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/syuant/Firmware/Tools/sitl_gazebo/models:/home/syuant/Firmware/Tools/sitl_gazebo/models:/home/syuant/Firmware/Tools/sitl_gazebo/models:/home/syuant/Firmware/Tools/sitl_gazebo/models:/home/syuant/Firmware/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/syuant/ros_workspace/devel/lib:/opt/ros/noetic/lib:/home/syuant/Firmware/build/px4_sitl_default/build_gazebo:/home/syuant/Firmware/build/px4_sitl_default/build_gazebo:/home/syuant/Firmware/build/px4_sitl_default/build_gazebo:/home/syuant/Firmware/build/px4_sitl_default/build_gazebo:/home/syuant/Firmware/build/px4_sitl_default/build_gazebo
```
When I tried launching the script `roslaunch package_px4 start.launch`, I still got this error
```
Resource not found: px4
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/home/syuant/ros_workspace/src/mavlink
ROS path [2]=/home/syuant/ros_workspace/src/mavros/libmavconn
ROS path [3]=/home/syuant/ros_workspace/src/mavros/mavros_msgs
ROS path [4]=/home/syuant/ros_workspace/src/mavros/mavros
ROS path [5]=/home/syuant/ros_workspace/src/mavros/mavros_extras
ROS path [6]=/home/syuant/ros_workspace/src/package_alpha
ROS path [7]=/home/syuant/ros_workspace/src/package_px4
ROS path [8]=/home/syuant/ros_workspace/src/mavros/test_mavros
ROS path [9]=/opt/ros/noetic/share
The traceback for the exception was written to the log file
```

Any ideas on why this is happening? Thanks in advance.

closed 2022-08-26 by fatburg3r

stevenchangg, 2022-10-19:
Hello @fatburg3r 
Did u manage to found a solution to this issue? Because I am facing the exact same problem rn 😅
Thanks in advance!

fatburg3r, 2022-10-21:
I didn't figure out what was wrong, I got it working on the second try but not on the third. Not sure if this will help, but here's the end of my .bashrc file.

`# for ros1 noetic
source /opt/ros/noetic/setup.bash
source ~/ros_workspace/devel/setup.bash

source ~/Firmware/Tools/setup_gazebo.bash ~/Firmware ~/Firmware/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/Firmware
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/Firmware/Tools/sitl_gazebo
export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/home/syuant/Firmware/Tools/sitl_gazebo/models`

AlexWUrobot, 2023-01-24:
I solved it by this way
https://discuss.px4.io/t/mavros-posix-sitl-launch-is-neither-a-launch-file-in-package-px4-nor-is-px4-a-launch-file-name/30522/2
https://docs.px4.io/main/en/simulation/ros_interface.html

AlexWUrobot, 2023-01-24:
I solved it by this way, based on Ubuntu 18.04 and Python 3.6
https://discuss.px4.io/t/mavros-posix-sitl-launch-is-neither-a-launch-file-in-package-px4-nor-is-px4-a-launch-file-name/30522
https://docs.px4.io/main/en/simulation/ros_interface.html

topiga, 2023-04-19:
Hey @fatburg3r, did you find a solution for this : https://discuss.px4.io/t/failed-to-use-mavlink-mav-cmd-set-message-interval/28921
I’m having exactly the same problem as you do