PX4/PX4-Autopilot #20118: Pixhawk 6C Telem1-3 not working as expected

ecmnet, 2022-08-27:
With Pixhawk 6C, Telem1-3 connections seem not to work properly. No connection to companion possible. MAVLink starts as expected, but serial connection fails (providing some garbage). Flow control is off, baudrate checked. Same HW (cable, companion) with Pixhawk4 and Cube Orange is working. Tried all ports with multiple baudrates. Any suggestions to solve this?
@dagar @davids5 @vincentpoont2

ryanjAA, 2022-08-27:
can you upload a log. My telem is working.

ecmnet, 2022-08-27:
[image]

ryanjAA, 2022-08-27:
can you just do a simple wireless telemetry connection and at a slower baud rate

ecmnet, 2022-08-27:
dmesg output:

[image]

ecmnet, 2022-08-27:
Got it working now with 57600 baud (this did not work before starting BARO and MAG on external I2C. Still 921600 should be possible without flow control...

ryanjAA, 2022-08-27:
Good to hear. 

Agreed. So we may need @davids5 to chime in here. 

@vincentpoont2 have you ever gotten it working at max baud speed? I haven’t tried it that fast but will do. 

@ecmnet i can see you’re on main, can you step it back about a month. Just choose a random commit to see if there is anything in there.

ecmnet, 2022-08-27:
One additional observation: 56700 baud only works for TELEM1 but not for TELEM2.

davids5, 2022-08-28:
> Good to hear.
> 
> Agreed. So we may need @davids5 to chime in here.
> 
> @vincentpoont2 have you ever gotten it working at max baud speed? I haven’t tried it that fast but will do.
> 
> @ecmnet i can see you’re on main, can you step it back ~about a month. Just choose a random commit~ just use release 1.13.0. I know a bunch of other things aren’t working in that but it will be the cleanest/most tested to see if there is anything in now inadvertently altering things.

My verification notes say that Telem 1 was tested to 1.5 mbs with no errors. Telem 2 did not have 0 errors at 3 mbps.  I would check cabling and grounds.

ecmnet, 2022-08-28:
@davids5 Connection works fine with other HW. So I do not expect any hardware related root causes (Twistet RX/TX with ground, verified that no ground loop exists). I again verified that TELEM1 is working up to 460800 but not with 921600 forcing flow control off.

ecmnet, 2022-08-31:
**Update:** TELEM2 is working up to 3MBit baud rate without any errors using flow control. Closing the issue as not relevant for SW. Obviously the serial signal of Pixhawk6C is not as clean as Pixhawk4 or CubeOrange (as it works for both up to 921600 baud without flow control).

@vincentpoont2 @davids5 @dagar

closed 2022-08-31 by ecmnet