PX4/PX4-Autopilot #20126: SITL Gazebo Iris Max Speed bug

Edekheh, 2022-08-29:
## Describe the bug
SITL max speed limitation for quadcopter to 20m/s. MPC_XY_VEL_MAX and MPC_XY_CRUISE doesn't change max speed.

## To Reproduce
Steps to reproduce the behavior:
1. Run roslaunch px4 mavros_posix_stl.launch
2. Open QGroundControl, change MPC_XY_VEL_MAX to 50m/s and MPC_XY_CRUISE to 40m/s
3. After sending drone to distant location you will se max speed at 20m/s instead of 40.

## Expected behavior
According to changing in parameters, flight controller should not limit SITL drone to 20m/s.

Add screenshots to help explain your problem.

## Drone (please complete the following information):
- SITL IRIS run by 
- Photo of the IMU / autopilot setup if possible.

## Additional context
Add any other context about the problem here.

andreykanevskoi, 2024-03-21:
Check [docs](https://docs.px4.io/v1.9.0/en/advanced_config/parameter_reference.html), this is the maximum allowed speed. It won't work faster, the speed will be reduced.
[image]

Edekheh, 2024-03-21:
Oh wow it was so long ago i forgot about it.
To modify speed above 20 m/s You can go to Tools/simulation/sitl-gazebo/src/gazebo_motor_model.cpp and edit line 222 : 
double scalar = 1 - vel / 25.0; // at 25 m/s the rotor will not produce any force anymore

Now You can fly up to 35-40 m/s.

If You want to fly even faster You need to modify simulated vehicle sdf model, so that rotors will produce more thrust :)

closed 2024-03-21 by Edekheh

andreykanevskoi, 2024-03-21:
I use AirSim and MavLink, couldn't found smth similar there ;(