PX4/PX4-Autopilot #20129: roll, pitch controls are not working properly when large yaw error introduced

limhyon, 2022-08-29:
## Describe the bug
It seems that controller is using yaw set point to generate body rate. But when large yaw error introduced, it makes roll, pitch controller not working.

## To Reproduce
Steps to reproduce the behavior:
1. Control roll, pitch with large yaw error.2. 
2. Roll, pitch will not be controlled. It seems reversed when 90 degrees yaw error exist. (See log)

## Expected behavior
Roll, pitch should controlled well.
- if you see log, roll pitch controllers are well tuned.
- but suddenly, after 2:05 of log, roll, pitch angle control is not working well.
- it seems roll, pitch controls are reversed.

## Log Files and Screenshots
https://logs.px4.io/plot_app?log=5f4941e7-27aa-4666-9341-6b588528e63f

MaEtUgR, 2022-08-30:
Hi @limhyon ! Nice to hear from you and thanks for the report.
I'd need to double check but what I'd expect is that if you have a large steady state yaw error then your drone is not alligned with the attitude setpoints heading and the setpoint's heading is what determines the stick mapping. So it's like steering a ghost setpoint with a certain heading and if your drone is not alligned in heading then your roll and pitch stick input are mapped such that the drone tilts like with that ghost which is the setpoint.

In the plots that also looks weird because we generate the plots from quaternions and the roll and pitch angles have a different meaning if yaw is off. So roll and pitch control should still work fine even though yaw is off but the plot doesn't show that correctly because they are coupeled. If roll and pitch control would really get out of wack then the drone would deviate immediately from level flight and crash.

Does that match your observations?

limhyon, 2022-09-04:
It seems that log does not reflect the set point that actually provided to the controller. For instance, when we download well flown session, roll/pitch are really off from set point. Does log really logging actual set point provided to the controller?

dagar, 2022-09-04:
Should we add something like this for stabilized mode as well? https://github.com/PX4/PX4-Autopilot/pull/18624

Something like `yaw_aligned` or `yaw_good_for_control` could be added to `vehicle_attitude` and then used in `MulticopterAttitudeControl::generate_attitude_setpoint()`.

dagar, 2022-09-04:
In mc_att_control we also have this bit of logic for large yaw errors, but it's only enabled for VTOL. Would it make sense to be available optionally for multicopters as well?

https://github.com/PX4/PX4-Autopilot/blob/7bbdc220f5dcb568f87020cf3837e21043f0a2ab/src/modules/mc_att_control/mc_att_control_main.cpp#L166-L177