PX4/PX4-Autopilot #20130: make px4_sitl none_iris target not found

Javit96, 2022-08-29:
## Describe the bug
Problem executing the following command: 'make px4_sitl_default none_iris'.

##Enviroment
.Windows 11 Pro build:22622.586
.WSL2 with Ubuntu 22.04.1 LTS
.Latest PX4-Autopilot commit

## To Reproduce
Steps to reproduce the behavior:
1. Open Ubuntu22.04 Terminal (WSL2)
2. Run 'bash Tools/setup/ubuntu.sh' command
3. export PX4_SIM_HOST_ADDR=xx.xx.xx.xx(IP of mi wsl machine)
4. See error

## Expected behavior
I expect to see a succesful build and see px4 running on ports.

[image]

## Additional context
I'm already run 'make clean', and delete the local folder from my pc and download again.

tkhamvilai, 2022-09-25:
just use none instead of none_iris

dagar, 2022-09-25:
What are you actually trying to do with `none_iris`? I've been gradually trying to untangle a lot of this stuff that didn't really make sense to me.

Now in the `main` branch you can simply run the PX4 binary directly if you want.

``` Console
$ make px4_sitl_default

$ ./build/px4_sitl_default/bin/px4

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [param] selected parameter default file parameters.bson
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
INFO  [simulator_sih] Simulation loop with 250 Hz (4000 us sim time interval)
INFO  [simulator_sih] Simulation with 1.0x speedup. Loop with (4000 us wall time interval)
INFO  [lockstep_scheduler] setting initial absolute time to 4000 us
INFO  [commander] LED: open /dev/led0 failed (22)
No autostart ID found
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2022-09-25/17_36_40.ulg	
INFO  [logger] Opened full log file: ./log/2022-09-25/17_36_40.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh>
```

closed 2022-10-11 by Javit96