PX4/PX4-Autopilot #20134: PX4 is not sending RTCMStream DroneCAN msgs

vincentpoont2, 2022-08-31:
## Describe the bug
PX4 is not sending RTCMStream DroneCAN msgs, which it should be sending when it gets GPS_RTCM_DATA MAVLink messages. Test with DroneCAN F9P RTK, used DroneCAN GUI tool and PX4 isn't sending them.

## To Reproduce
Using a DroneCAN F9P RTK, unable to get RTK GPS Lock. 
[image]

Repeat the same with a i2c F9P RTK, no issue
[image]

AlexKlimaj, 2022-09-02:
Enable UAVCAN_PUB_RTCM
https://github.com/PX4/PX4-Autopilot/blob/main/src/drivers/uavcan/uavcan_params.c#L207

OfficerJannis, 2022-09-06:
I have the same problem. I tested the issue with an _Here3 GPS_ in CAN mode, aswell as in serial mode: 

In CAN mode the GPS lock does **not** switch to RTK (float), although i was sending RTCM-Date via MavLink. 
When i open the CLI, the `uavcan status` printout _gnss rtcm publisher counter_ increases.  

In serial mode the GPS lock switches immediately to RTK (float) if i send RTCM-Data via MavLink.

I tested this with the latest stable PX4 1.13.
The parameter UAVCAN_PUB_RTCM mentioned by @AlexKlimaj  is not in the latest stable build.

Then i also tested the latest developer build (master) which has the parameter UAVCAN_PUB_RTCM included. I also enabled it.
In CAN mode i don't get RTK and the the `uavcan status` printout _gnss rtcm publisher counter_ stops at 1 event.
I also enabled the UAVCAN_PUB_MBD parameter, still no RTK.

AlexKlimaj, 2022-09-06:
UAVCAN_PUB_RTCM is currently broken in main. See the fix here https://github.com/PX4/PX4-Autopilot/pull/20056

AlexKlimaj, 2022-09-15:
Fixed with https://github.com/PX4/PX4-Autopilot/pull/20056

closed 2022-09-15 by AlexKlimaj