PX4/PX4-Autopilot #20142: QGC Parameters missing in HITL

tayyabkhalil-313, 2022-09-01:
## Describe the bug
I am trying to setup my environment for HITL with pixhawk 5x/pixhawk 2 cube/pixhawk 1. Upon uploading the firmware (v1.13.0), QGC gives the error of missing parameters: "Parameters are missing from firmware. You may be running a version of firmware QGC does not work correctly with or your firmware has a bug in it. Missing params: -1:SYS_AUTOSTART, -1:SYS_AUTOCONFIG." 

I saved the parameters and then manually added them to the parameters file. But on uploading them, QGC is unable to detect the new parameters (SYS_AUTOSTART and SYS_AUTOCONFIG) although I am able to add another parameter (RC_MAP_ROLL) through this method. 

I have tried on both QGC v4.2.3 and v4.2.0.

vibes99, 2022-09-29:
Hi, I am also running through the same problems. I am using pixhawk 4 mini and upon uploading px4 firmware (v1.13.0), QGC (v4.2.3) tells me that I'm missing parameters (same error as above). Has anyone got any fix for this? I've tried with various combinations of px4 firmware and qgc releases and yield the same error. Is this something to do with the firmware like a bug, or do I have to change the hardware as it works perfectly well when I flash the same firmware with pixhawk 4 controller (instead of pixhawk 4 mini) and I can get all the parameters on QGC v4.2.3

tayyabkhalil-313, 2022-09-29:
Hi, 
I managed to solve this by formatting my SD Card and now it works. Try that!