PX4/PX4-Autopilot #20150: The way to get motor speed(motor rpm) in SITL

r135792uuuu, 2022-09-02:
Hello developers of px4! I'm new in px4 develop.
I meet a question about how to get the motor speed in SITL.
At first I tried to use dshot to get the motor speed, but I found that SITL seems not to support dshot ! 
**(the plane model which I use is iris, enviroments which are gazebo—windy.world and jmavsim—default)** 

So I tried to find another way to reply the motor speed. 
Such as:
`actuator_outputs_s output;`
`Omega(0) = (output.output[0] + 1 ) / 2;`
 which Omega(0) is the fist motor speed after normalization  (the same way to calculate last three motors).

Does the way to calculate motor speed right  or  is there other way to calculate motor speed in SITL?

r135792uuuu, 2025-06-26:
I also discovered this topic in the motor model plugin of px4-gazebo. This topic outputs the simulated motor speed. However, this topic is an internal one of gazebo and has not been posted in ROS. So I tried to forward this topic to ROS, but I found a problem. Since my px4 version is 1.12.3, once this topic is published in ROS, it will block the transmission of other topics. Therefore, I failed to forward it to ROS. I indexed the motor topic of gazebo. In protobuf definition, I found the author's note: "Forwarding ROS this topic will cause blocking." This was consistent with the result of my code, so I gave up the attempt in version 1.12.3. However, there are other methods, which are to transmit these data through some inter-process communication methods. Perhaps you can try a higher version of PX4 to see if it solves this forwarding problem.

YiyangMa8, 2025-06-26:
> [@r135792uuuu](https://github.com/r135792uuuu) [@YiyangMa8](https://github.com/YiyangMa8) You should write a gazebo plugin that subscribes to the motor velocities in gz messages, and republished it in ROS.

OK,thank you！I'll try it.

YiyangMa8, 2025-06-26:
> I also discovered this topic in the motor model plugin of px4-gazebo. This topic outputs the simulated motor speed. However, this topic is an internal one of gazebo and has not been posted in ROS. [...]

OK , thank you!

closed 2025-06-26 by r135792uuuu