PX4/PX4-Autopilot #20157: Fail compiling in Ubuntu 16.04

AllenHuangDongyue, 2022-09-03:
When I am trying to compile the program of v1.11.0 under Ubuntu 16.04,
I was following the code:
git clone https://github.com/PX4/Firmware.git --recursive

Then it shows the error:
fatal: reference is not a tree: fd12483ddd4e58242d61d74a163e7aeaa1e0f466
Unable to checkout 'fd12483ddd4e58242d61d74a163e7aeaa1e0f466' in submodule path 'src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan/dsdl'
Failed to recurse into submodule path 'src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan'
....
Failed to recurse into submodule path 'src/drivers/uavcan/libuavcan'

Is there anything I did wrong? Any suggestions will be welcome!

Regards, 
Allen

AllenHuangDongyue, 2022-09-03:
When I tried the suggestions posted in https://github.com/PX4/PX4-Autopilot/issues/6668,

Unfortunately, it still can't work, it shows:

fatal: reference is not a tree: fd12483ddd4e58242d61d74a163e7aeaa1e0f466
Unable to checkout 'fd12483ddd4e58242d61d74a163e7aeaa1e0f466' in submodule path 'src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan/dsdl'
Failed to recurse into submodule path 'src/drivers/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan'
Failed to recurse into submodule path 'src/drivers/uavcan/libuavcan'

AllenHuangDongyue, 2022-09-03:
I followed the same procedure to compile the program many months ago, and it works well. But it can't work right now.

Jaeyoung-Lim, 2022-09-03:
@AllenHuangDongyue Unfortunately Ubuntu 16.04 has reached end if life in April 2021, therefore it would be hard to support your issue

julianoes, 2022-09-03:
The commit it's trying to check out is no longer available.

If you don't need UAVCAN, you can disable building it, and just not check it out. For instance for fmu-v4 you would disable it here: https://github.com/PX4/PX4-Autopilot/blob/release/1.10/boards/px4/fmu-v4/default.cmake#L47 by commenting out the line.

If you do need it, you will have to find a commit of that submodule that works for you and check it out manually. It's hard to keep supporting these things for years back, so I strongly recommend using something more recent like v1.13, otherwise you are on your own and need to know how to deal with these problems, unfortunately.

AllenHuangDongyue, 2022-09-03:
> 

Got it, thanks!

closed 2022-09-03 by AllenHuangDongyue

AllenHuangDongyue, 2022-09-03:
> If you do need it, you will have to find a commit of that submodule that works for you and check it out manually.
BTW, could you tell me where I can add the new commit?

Many thanks!