PX4/PX4-Autopilot #20158: no_sim=1 command ends up opening up a gazebo simulator

returncode13, 2022-09-03:
## Describe the bug
I am trying to launch px4 sitl while launching gazebo separately via ros.  
Expect
    no_sim=1  make   px4_sitl_default   gazebo   
to start px4 sitl without launching a gazebo simulator
But I find that this launches the simulator ignoring  the no_sim=1 flag.

## To Reproduce
Steps to reproduce the behavior:
1. clone this repo
2. run the command  " no_sim=1  make  px4_sitl_default  gazebo "

## Expected behavior
The px4 sitl coming up **without** a simulator

dagar, 2022-09-03:
At the moment in the main branch simulation is in flux and we need to do a sweep to gets the docs back in sync.

If currently on `main` you can now simply run the px4 binary directly after compiling.

``` sh
$ make px4_sitl_default
$ ./build/bin/px4
```

Within PX4 running you can set the SYS_AUTOSTART parameter or you you can pass an environment variable PX4_SYS_AUTOSTART=4001, etc.

Please let me know you're full use case and I'll try to make it as streamlined as possible, I'm currently focused on new Gazebo (formally known as Ignition Gazebo).

returncode13, 2022-09-04:
Hi @dagar . thank you for the workaround. The use case is a simple learning exercise to separate px4 and have gazebo open as part of ros launch . I am now presented with a new issue in which the gazebo launched via ros isn't connected to px4. 
Steps were as follows.
1. launch px4 sitl as per your workaround
2. export GAZEBO_MODEL_PATH, GAZEBO_PLUGIN_PATH
3. sourced ros
4. ros launch in a new terminal which opens up gazebo

I expect the sitl to connect with the gazebo simulator once I load the iris model via gazebo.
But I find thats not the case, sitl still reports 
"INFO  [simulator_mavlink] Waiting for simulator to accept connection on TCP port 4560"

returncode13, 2022-09-04:
Hi @dagar . The use case is a simple learning exercise to separate px4 and have gazebo open as part of ros launch, insert a model and have the sitl connect with gazebo . Thank you for the workaround. Closing this issue.

closed 2022-09-04 by returncode13

dagar, 2022-09-04:
Thanks for the response, let me know how it goes. We'll also have much deeper integration with PX4 in ROS2 soon.