PX4/PX4-Autopilot #20187: Cube orange reboots when micrortps_client is started

AjayaDahal, 2022-09-10:
I think I have successfully flashed rtps firmware in my cube orange. But please let me confirm. Here is what I did. Initially, I got error somewhere along this line saying memory full (102%) when I tried adding the following four lines in the default.px4board. So, I commented #CONFIG_DRIVERS_CAMERA_CAPTURE=y and #CONFIG_DRIVERS_CAMERA_TRIGGER=y in cubeorange's default.px4board. In addition to that, I created rtps.px4board and moved these lines from default.px4board which I had added earlier:

CONFIG_DRIVERS_HEATER=n
CONFIG_DRIVERS_OSD=n
CONFIG_MODULES_MICRODDS_CLIENT=y
CONFIG_MODULES_MICRORTPS_BRIDGE=y

I was able to compile it successfully and upload to the board. I have connected TELEM2 with my companion computer (Jetson nano). When I run micrortps_agent and sensor combined listener node in the Jetson, I dont see anything and I realized it was because the micrortps_client was not running in orange cube. So, I opened QGroundControl and opened Mavlink console to start the micrortps_client but when I run micrortps_client start -d /dev/ttyS2, the cube restarts. Thank you for the help in advance.

dagar, 2022-09-10:
Do you have anything else already running on TELEM2? Try enabling micrortps via parameter (RTPS_CONFIG).

AjayaDahal, 2022-09-10:
I did but the cube keeps restarting.

beniaminopozzan, 2022-09-11:
Micrortps_client and microdds don't work together (I don't know why). So just remove the microdds_client module 
`CONFIG_MODULES_MICRODDS_CLIENT=n`
And it should work. See here
https://github.com/PX4/PX4-Autopilot/issues/19852#issuecomment-1172370641
(It's not necessary to delete the folder)

PR #20202 (2022-09-15): remove microdds from rtps target
## Describe problem solved by this pull request
Fixes #20187 and #19852

## Describe your solution
Remove the `microdds_client` module for `rtps` targets as it causes problems.

## Describe possible alternatives
Maybe create separate build configurations for enabling Micro XRCE-DDS?

## Test data / coverage
The issue is verified on my own hardware. I tested this on a pixracer running `px4_fmu-v4_rtps` firmware.
closed 2022-09-15 by dagar