PX4/PX4-Autopilot #20211: PX4IO not processing RC input

davids5, 2022-09-14:
## Describe the bug
While tracing down a HW bug - I could not test DSM SBUS or PPM on V6X HW

## To Reproduce
Steps to reproduce the behavior:

1. Connect fmu v6x
2. Load main
3. param set 4001
4.  param set RC_INPUT_PROTO -1
5. connect any of PPM SBUS or DSM
6. px4io status

Note Slow response and `0 raw R/C inputs`

## Expected behavior

AlexKlimaj, 2022-09-14:
I can confirm SBUS RC input is not working on the V6X and V5X with current main. It works on 1.13.

AlexKlimaj, 2022-09-14:
It looks like it is failing here.
https://github.com/PX4/PX4-Autopilot/blob/main/platforms/nuttx/src/px4/stm/stm32h7/px4io_serial/px4io_serial.cpp#L361

That error propagates up and results in this failing.
https://github.com/PX4/PX4-Autopilot/blob/main/src/drivers/px4io/px4io_serial.cpp#L179

Then that results in this failing.
https://github.com/PX4/PX4-Autopilot/blob/ce337a3d80e71032511b77251596e2f1ec4681b4/src/drivers/px4io/px4io.cpp#L1187

Ultimately the io_get_status() fails.
https://github.com/PX4/PX4-Autopilot/blob/main/src/drivers/px4io/px4io.cpp#L925

davids5, 2022-09-14:
That is a timeout on the DMA completion - it has to be traced to see the 3 possible cause

AlexKlimaj, 2022-09-14:
Weird the initial io_reg_get in init are working. But then fail in run().