PX4/PX4-Autopilot #20219: No mixer file found in the current branch of PX4 but I can found in V1.10

Jawad-RoboLearn, 2022-09-15:
No mixer file found in the current branch of PX4 but I can found in V1.10

The default location is src/lib/mixer but cannot found in the current branch.

mcsauder, 2022-09-15:
Hi @Jawad-RoboLearn , the flight stack has migrated from the old mixer system to the control allocation system.  You will still find mixers present in v1.13.0, the latest stable release.

Here is that change: https://github.com/PX4/PX4-Autopilot/pull/20108

Here are docs: https://docs.px4.io/main/en/concept/control_allocation.html

You can find the old mixer file present in v1.13.0 in the tagged release: https://github.com/PX4/PX4-Autopilot/tree/6823cbc4140e29568f00e1211ae60e057adb1a1f/src/lib/mixer
Also, airframe specific files here in the last release: https://github.com/PX4/PX4-Autopilot/tree/6823cbc4140e29568f00e1211ae60e057adb1a1f/ROMFS/px4fmu_common/mixers

If this answers your question feel free to close this issue!

Jawad-RoboLearn, 2022-09-15:
Dear Sauder
Ok, i will find it.

closed 2022-09-20 by bresch