PX4/PX4-Autopilot #20229: make file to do reply for ekf2

tahan094, 2022-09-16:
## Describe the bug
Following the guide here https://docs.px4.io/main/en/debug/system_wide_replay.html, I get the issue when calling "make px4_sitl none"
"[0/2] Re-checking globbed directories...
ninja: error: unknown target 'none', did you mean 'tunes'?
make: *** [Makefile:237: px4_sitl] Error 1"

## To Reproduce
Steps to reproduce the behavior:
use the guideline for replaying ekf2 after export the mode and the ulg file

## Expected behavior
to have the replay working and to get 
"INFO  [replay] Replay done (published 9917 msgs, 2.136 s)"
also to have the following path created /PX4-Autopilot/build/px4_sitl_default_replay/tmp/rootfs/replay_params.txt

## Log Files and Screenshots
NA, only using vs code and the guide u provided

Add screenshots to help explain your problem.

## Drone (please complete the following information):
NA

## Additional context
NA

dagar, 2022-09-16:
This is a regression I'll fix shortly.

tahan094, 2022-09-16:
Can you send me the fix after you do or tell me how to get it?

tahan094, 2022-09-16:
@dagar do you have an estimate for this fix? I really need it asap

tahan094, 2022-09-17:
@dagar hello?!

tahan094, 2022-09-17:
@dagar I tried to checkout v1.12.3 and to do the replay but it seems that the structure was not the same between it an 1.13 as it gives error that time stamp could not be read??,  and I really need to use 1.13 as it includes the NMEA and it is needed at my side for specific GPS Can not be changed

tahan094, 2022-09-19:
@dagar hello, u have an estimate for the change please answer me?

tahan094, 2022-09-21:
@dagar hello why do you ignore me?

dagar, 2022-09-22:
Please try https://github.com/PX4/PX4-Autopilot/pull/20282.

tahan094, 2022-09-23:
Hello,  it is worked but has a lot issues as you can see below 

Other issue related to pyulog
make px4_sitl none
[0/2] Re-checking globbed directories...
[0/1] cd /home/tahan094/PX4-Autopilot/build/px4_sitl_default_replay/rootfs && /home/tahan094/PX4-Autopilot/build/px4_sitl_default_replay/bin/px4
INFO  [uORB] Using orb rules from ./orb_publisher.rules

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SDLOG_PROFILE: curr: 1 -> new: 3
INFO  [replay] using replay log file: /home/tahan094/Desktop/Fly5/log_7_Ù¢Ù Ù§Ù -Ù¡-Ù¡-Ù Ù¢-Ù¥Ù§-Ù¢Ù¦.ulg
INFO  [replay] Applying params from ULog file...
INFO  [replay] Applying override params from ./replay_params.txt...
INFO  [ekf2] replay mode enabled
WARN  [ekf2] Use EKF2_GPS_CTRL instead

INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2022-09-23/11_30_05_replayed.ulg
INFO  [logger] Opened full log file: ./log/2022-09-23/11_30_05_replayed.ulg
INFO  [replay] Ekf2 replay mode
INFO  [replay] Applying params from ULog file...
INFO  [replay] Applying override params from ./replay_params.txt...
INFO  [replay] Replay in progress...
ERROR [replay] Formats for actuator_armed don't match. Will ignore it.
WARN  [replay]  Internal format:
WARN  [replay]  - uint64_t timestamp
WARN  [replay]  - bool armed
WARN  [replay]  - bool prearmed
WARN  [replay]  - bool ready_to_arm
WARN  [replay]  - bool lockdown
WARN  [replay]  - bool manual_lockdown
WARN  [replay]  - bool force_failsafe
WARN  [replay]  - bool in_esc_calibration_mode
ERROR [replay]  - uint8_t[1] _padding0
WARN  [replay]  File format    : uint64_t timestamp;bool armed;bool prearmed;bool ready_to_arm;bool lockdow
WARN  [replay]  - uint64_t timestamp
WARN  [replay]  - bool armed
WARN  [replay]  - bool prearmed
WARN  [replay]  - bool ready_to_arm
WARN  [replay]  - bool lockdown
WARN  [replay]  - bool manual_lockdown
WARN  [replay]  - bool force_failsafe
WARN  [replay]  - bool in_esc_calibration_mode
ERROR [replay]  - bool soft_stop
ERROR [replay] Formats for actuator_controls_0 don't match. Will ignore it.
WARN  [replay]  Internal format:
WARN  [replay]  - uint64_t timestamp
WARN  [replay]  - uint64_t timestamp_sample
ERROR [replay]  - float[9] control
ERROR [replay]  - uint8_t[4] _padding0
WARN  [replay]  File format    : uint64_t timestamp;uint64_t timestamp_sample;float[8] control;
WARN  [replay]  - uint64_t timestamp
WARN  [replay]  - uint64_t timestamp_sample
ERROR [replay]  - float[8] control
WARN  [replay] Topic actuator_controls_3 not found internally. Will ignore it
ERROR [replay] Formats for failure_detector_status don't match. Will ignore it.
WARN  [replay]  Internal format:
WARN  [replay]  - uint64_t timestamp
WARN  [replay]  - float imbalanced_prop_metric
[... 1130 lines cut ...]
ERROR [replay] Dropout in replayed log, 3 ms
ERROR [replay] Dropout in replayed log, 202 ms
ERROR [replay] Dropout in replayed log, 206 ms
ERROR [replay] Dropout in replayed log, 182 ms
ERROR [replay] Dropout in replayed log, 17 ms
ERROR [replay] Dropout in replayed log, 0 ms
ERROR [replay] Dropout in replayed log, 188 ms
ERROR [replay] Dropout in replayed log, 3 ms
ERROR [replay] Dropout in replayed log, 120 ms
ERROR [replay] Dropout in replayed log, 129 ms
ERROR [replay] Dropout in replayed log, 0 ms
ERROR [replay] Dropout in replayed log, 293 ms
ERROR [replay] Dropout in replayed log, 3 ms
ERROR [replay] Dropout in replayed log, 192 ms
ERROR [replay] Dropout in replayed log, 234 ms
ERROR [replay] Dropout in replayed log, 0 ms
ERROR [replay] Dropout in replayed log, 93 ms
ERROR [replay] Dropout in replayed log, 160 ms
INFO  [replay] Replay done (published 31 msgs, 0.773 s)
INFO  [replay] 
INFO  [replay] Topic, Num Published, Num Error (no timestamp match found):
INFO  [replay] vehicle_air_data: 409, 2
INFO  [replay] vehicle_magnetometer: 275, 2
INFO  [logger] closed logfile, bytes written: 111105
Powering off NOW.Exiting NOW.

tahan094, 2022-09-23:
Pyulog issue
tahan094@tahan094:~/PX4-Autopilot$ ulog_params -i "$replay" -d ' ' | grep -e '^EKF2' > build/px4_sitl_default_replay/rootfs/replay_params.txt Traceback (most recent call last): File "/home/tahan094/.local/bin/ulog_params", line 8, in <module> sys.exit(main()) File "/home/tahan094/.local/lib/python3.8/site-packages/pyulog/params.py", line 62, in main params = get_defaults(ulog, args.default) File "/home/tahan094/.local/lib/python3.8/site-packages/pyulog/params.py", line 20, in get_defaults raise Exception('invalid value \'{}\' for --default'.format(default)) Exception: invalid value ' ' for --default

dagar, 2022-09-23:
You need to replay with the exact same version of PX4 (or at least the .msg files must match).

tahan094, 2022-09-24:
Can you please clarify what to write in the command, I am using px4 fmu v3
Normally I used these commands as described [here](https://docs.px4.io/main/en/debug/system_wide_replay.html)

export replay_mode=ekf2
export replay=<abs_path_to_log.ulg>
make px4_sitl none

tahan094, 2022-10-05:
Hello @dagar I will open this as an issue