PX4/PX4-Autopilot #20248: Supress disarm in manual mode (only for Fixed Wing)

jstrebel, 2022-09-19:
**Describe the enhancement:**
To test the flight behavior of new fixed wing UAVs it may be necessary to perform flight maneuvers in manual mode that result in a "disarm" in the current XP4 software. This type of flight movement may be required to put the FW into critical flight conditions to determine how the aircraft is behaving.
Currently, remote control is no longer possible, it results in a crash.

**To Reproduce:**
Put the FW upside down (180° Roll). FW in going to disarm.
 
**Expected behavior**
Insert an optional parameter so that disarm can be suppressed.

ryanjAA, 2022-09-27:
I thought it wouldn’t disarm if the in air flag was valid.

jstrebel, 2022-09-27:
I can not confirm this.

dagar, 2022-09-27:
In the current code failure detector max roll/pitch should only be active in an attitude control mode. That means for FW manual and acro it shouldn't have triggered.

https://github.com/PX4/PX4-Autopilot/blob/0ee4d89744c7719849277befbeadc4581b2d208b/src/modules/commander/failure_detector/FailureDetector.cpp#L173-L174

What version of PX4 exactly? Was there any log to recover?

jstrebel, 2022-09-29:
It was 1.13.0

ryanjAA, 2022-09-29:
If I have throttle down and am in manual mode and yaw all the way to the left will this disarm? That's also relevant as well here.

jstrebel, 2022-09-29:
I think its relevant but not so important you have to remain yaw left for some seconds. May be we should put a time window (may be there is already one). It is very practical especially during setup on the bench to arm or disarm via rc stick. In actual flying there are two maneuvers where you put the yaw rudder for a few seconds. Renversement or Knive

ryanjAA, 2022-09-30:
Think about landing in a crosswind - it is important as you could definitely do it. This was actually brought to my attention by a long time flyer. A parameter is the best option for this.

jstrebel, 2022-10-11:
Could we consider to ARM or disARM using a separate RC channel?

jstrebel, 2022-10-12:
Hi,
here is the log file as promised during the conf call yesterday.
Jakob
Your uploaded log file is available under:
https://review.px4.io/plot_app?log=fa954594-b698-49fb-a00e-c21ac760d0e9