PX4/PX4-Autopilot #20249: Code formatting with `clang-format`

howjmay, 2022-09-19:
## Describe problem solved by the proposed feature
clang-format is a more generally used formatting tool. Its installation is easier than astyle

## Describe your preferred solution
A clear and concise description of what you want to happen.

## Describe possible alternatives
using clang-format!

tstastny, 2022-09-20:
@howjmay, @MaEtUgR has started this process in this PR https://github.com/PX4/PX4-Autopilot/pull/19855

Maybe you could help support with testing the PR / working on the formatting?

isidroas, 2024-06-17:
We could take inspiration from https://github.com/git/git/blob/master/Makefile

They use https://clang.llvm.org/docs/ClangFormat.html#git-integration

```
  To format staged changes, i.e everything that's been `git add`ed:
    git clang-format

  To also format everything touched in the most recent commit:
    git clang-format HEAD~1

  If you're on a branch off main, to format everything touched on your branch:
    git clang-format main
```

This is shiped in the `clang-format` Ubuntu/debian package.

It formats regions of changed code, not the entire file, so the migration is more progressive.