PX4/PX4-Autopilot #20251: MAVLink tx rate mult is set to a much lower value than the documentation describes for a given MAV_x_RATE setting

logos-electromechanical, 2022-09-19:
## Describe the bug
The output rate of mavlink messages on serial channels is far less than I would expect for a given baud rate and MAV_x_RATE setting. Specifically, the tx rate mult reported by mavlink status is *MUCH* lower than required to prevent overflowing channel bandwidth.

## To Reproduce
Steps to reproduce the behavior:
1. Set the desired baudrate of a serial telemetry connection to a high value (for example, 921600)
2. Set the corresponding MAV_x_MODE to a high rate mode (for example, Onboard)
3. Set the corresponding MAV_x_RATE to baudrate/20 (46080 matches the numbers above)
4. Enter 'mavlink status streams' into the mavlink console and observe that the transmitted rates of messages on the channel previously configured are far below default. 
5. Enter 'mavlink status' into the mavlink console and observe that the tx rate mult on the configured MAVLink channel is much less than 1.000. I observed 0.050 with the settings above. 
6. If you set MAV_x_RATE to 921600 or higher with the other settings the same, you should observe the mavlink messages coming out at full rate and a reported transmit bandwidth far below the capacity of the interface. 

## Expected behavior
I expect that I will get all messages at full rate with the example settings given above. With the workaround described to get full rate, I'm seeing actual transmit bandwidth below 7000 bytes/sec.

## Log Files and Screenshots
Here's the outputs for mavlink status, with channels set to demonstrate both the problem and the workaround:
```
instance #0:
    mavlink chan: #0
    type:        GENERIC LINK OR RADIO
    flow control: OFF
    rates:
      tx: 6836.1 B/s
      txerr: 0.0 B/s
      tx rate mult: 1.000
      tx rate max: 921600 B/s
      rx: 0.0 B/s
      rx loss: 0.0%
    FTP enabled: YES, TX enabled: YES
    mode: Onboard
    MAVLink version: 2
    transport protocol: serial (/dev/ttyS6 @460800)

instance #1:
    mavlink chan: #1
    type:        GENERIC LINK OR RADIO
    flow control: OFF
    rates:
      tx: 865.9 B/s
      txerr: 0.0 B/s
      tx rate mult: 0.050
      tx rate max: 46080 B/s
      rx: 0.0 B/s
      rx loss: 0.0%
    FTP enabled: YES, TX enabled: YES
    mode: Onboard
    MAVLink version: 2
    transport protocol: serial (/dev/ttyS4 @230400)

instance #2:
    mavlink chan: #2
    type:        GENERIC LINK OR RADIO
    flow control: OFF
    rates:
      tx: 790.5 B/s
      txerr: 0.0 B/s
      tx rate mult: 0.050
      tx rate max: 23040 B/s
      rx: 0.0 B/s
      rx loss: 0.0%
    FTP enabled: YES, TX enabled: YES
    mode: Onboard
    MAVLink version: 2
    transport protocol: serial (/dev/ttyS1 @921600)
```

## Drone (please complete the following information):
Pixhawk 6X. It's currently not connected to any other hardware beyond the GPS it came with -- I'm doing some development for a sail-powered USV, but AFAICT I haven't touched anything that should affect this.

dagar, 2022-09-20:
Thanks for reporting, we need to review the date rate calculation logic in detail. I think in a lot of cases this might go unnoticed on serial with flow control connected.

In the meantime do you have everything you need?

logos-electromechanical, 2022-09-20:
Yes -- my particular problem is taken care of b/c I have a usable workaround.