PX4/PX4-Autopilot #20277: EKF2 attitude drift when no position source available

dagar, 2022-09-21:
https://logs.px4.io/plot_app?log=04c4fe9e-0cf6-46cb-9528-1050f5546afa

 - ekf2 with no position source, after holding a sustained pitch forward an error in the attitude estimate develops
[image]
 - after updating accel bias inhibit logic (https://github.com/PX4/PX4-Autopilot/pull/20224) for this case it's slightly improved, but still a problem now showing up in the gyro bias estimate

[image]

priseborough, 2022-09-29:
@dagar I tried replaying the log using upstream master at commit cb2738e , but was unable to do so with the following errors

[image]

If we can get the replay working, then fixing this issue will be much faster.

priseborough, 2022-09-29:
Inspection of the code shows it is using an unrealistically small observation noise for the fake position fusion when in the air of 0.5m 1-sigma and 0.01m when on ground. For reference ArduPilot uses 10m when  in air and 0.5m when on ground.

dagar, 2022-09-29:
When flying (or armed on the ground) it should be 10 m (EKF2_NOAID_NOISE, unless EKF2_GPS_P_NOISE is larger).

https://github.com/PX4/PX4-Autopilot/blob/f89df9d9865386c93891cea55d2848139928489d/src/modules/ekf2/EKF/fake_pos_control.cpp#L120-L121

When disarmed/not in air it's 0.5m, and only goes to 0.01m if we're fairly confident the vehicle is still (sensitive IMU checks).

I'll take a look at the current replay situation and get you something usable.

danielmellinger, 2022-11-08:
Hi @priseborough, it would be awesome if you could provide some information on the design of EKF2 regarding this issue.  I am working with a multirotor vehicle that does not have a direct estimate of x or y position or velocity (that is, no GPS or vision sensor).  In running the SITL test described above I can see that the “fake position" feature continues to use static x and y position observations of 0 even when the vehicle has traveled a long distance from this point.  This large incorrect input into EKF2 seems to lead to the errors in the attitude estimate that show up in the above plot.  Disabling the fake position fusion feature does eliminate the large attitude estimation error in the above SITL test.  However, in simple indoor hover tests with a real vehicle with the fake position feature disabled attitude bias errors seem to develop and the vehicle flies as if it is poorly trimmed.

So, for a vehicle without an x or y position or velocity estimate, what feature(s) of EKF2 enable it to estimate an unbiased estimate of the vehicle attitude?  It seems the “fake position” feature attempts to achieve this goal but has issues in scenarios where the fake static position assumption is not valid.

Thank you so much!

Donecle, 2022-11-15:
Hi all,

A few years ago, we were using a very modified version of PX4's EKF2, basically had removed all what we didn't need/understand. I don't remember if fake position mechanism was already present but if it was we removed it.
Logically we had the behavior described above (drift in attitude) when no horizontal position/velocity was fused.
We solved the problem by fusing an estimate of true roll & pitch based on low-pass filtered accel measurements. The idea was to copy the Attitude Q estimator's behavior. The functions to fuse roll and pitch are not implermented in the standard EKF2 so we had to compute them, inspired by what is done on yaw. It worked pretty well in gentle conditions - indoor flight with no long-period dynamics.
I believe this would solve the described issue here, but would introduce another one, which is long-term attitude alignment with accel measurments, this would be bad for a fiwed-wing aircraft loitering in circles for instance.
Maybe it is possible to switch smartly between this accel-estimated roll&pitch fusion and fake position fusion...
Unfortunately we now are happy with fake position, so I do not have up to date code for this.

Hope this can help !

PR #21038 (2023-02-07): Pr ekf2 gravity observation
### Solves Problem
Fixes #20277 

This adds a [gravity observation](https://journals.sagepub.com/doi/10.5772/58463#sec-2-4-1) to `EKF2` based on the recommendations in [this thread](https://discord.com/channels/1022170275984457759/1041806515071361034).

### Outstanding Issues
It is unclear if this observation will play nicely with the Fake Position Fusion. It's also unclear whether the IMU bias states are observable in situations without a true position observation like GPS. In our testing we've disabled all of the above.

This was forward-ported from a 1.13 version [... cut]
closed 2023-02-07 by dagar