PX4/PX4-Autopilot #20295: Holybro Kakute H7, it can't use DShot as ESC protocol

helloliver, 2022-09-23:
Hello,

I use Holybro Kakute H7. I want to use DShot instead of PWM. But after I config the DShot, the motor cannot spin anyway. Besides, the official doc [dshot-escs](https://docs.px4.io/main/en/peripherals/dshot.html?#dshot-escs) seems help nothing.

Is there way to use DShot as ESC protocol for Holybro Kakute H7?

Thanks.

mywisdomfly, 2022-09-28:
I'm using Kakute H7 (old version MPU6000). Dshot is fine to me.

helloliver, 2022-09-28:
@mywisdomfly I would appreciate it if you can provide the PX4-Autopilot commit ID that you used.

mywisdomfly, 2022-09-28:
I'm using the latest release firmware and QGC.

mywisdomfly, 2022-10-20:
> @mywisdomfly Can you switch into offborad mode with Kakute H7? Thanks!
> 
> > I'm using Kakute H7 (old version MPU6000). Dshot is fine to me.

Yes.

julianoes, 2023-01-06:
I'm going to close this issue as a duplicate of https://github.com/PX4/PX4-Autopilot/issues/20345. I'm aware that this issue was first but the other one has more information.

And fix should be coming in: https://github.com/PX4/PX4-Autopilot/pull/20873

closed 2023-01-06 by julianoes