PX4/PX4-Autopilot #20302: make px4_sitl jmavsim GUI failed

tkhamvilai, 2022-09-25:
## Describe the bug
Cannot start jmavsim on WIndows 11 under WSL2 environment.

## To Reproduce
Follow the official instruction: [https://docs.px4.io/main/en/dev_setup/dev_env_windows_wsl.html#windows-10-gui-support](url)

## Expected behavior
A clear and concise description of what you expected to happen.

## Log Files and Screenshots
`make px4_sitl_default jmavsim
[0/2] Re-checking globbed directories...
[0/1] cd /home/tkhamvilai/PX4-Autopilot/build/px4_sitl_default...utopilot /home/tkhamvilai/PX4-Autopilot/build/px4_sitl_defaul
SITL ARGS
sitl_bin: /home/tkhamvilai/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
src_path: /home/tkhamvilai/PX4-Autopilot
build_path: /home/tkhamvilai/PX4-Autopilot/build/px4_sitl_default
SITL COMMAND: "/home/tkhamvilai/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/home/tkhamvilai/PX4-Autopilot/build/px4_sitl_default"/etc

______  __   __    ___
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 330 bytes, decoded 330 bytes (INT32:10, FLOAT:6)
INFO  [param] selected parameter backup file parameters_backup.bson
  SYS_AUTOCONFIG: curr: 0 -> new: 1
Buildfile: /home/tkhamvilai/PX4-Autopilot/Tools/simulation/jmavsim/jMAVSim/build.xml
  SYS_AUTOSTART: curr: 0 -> new: 10016
* SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
* SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by javax.media.j3d.JoglPipeline (rsrc:j3dcore.jar) to method sun.awt.AppContext.getAppContext()
WARNING: Please consider reporting this to the maintainers of javax.media.j3d.JoglPipeline
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
PX4 SIM HOST: localhost
INFO  [simulator_mavlink] Waiting for simulator to accept connection on TCP port 4560
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:567)
        at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:61)
Caused by: java.lang.InternalError: XXX0 profile[2]: GL2 -> profileImpl GL3bc !!! not mapped
        at com.jogamp.opengl.GLProfile.computeProfileMap(GLProfile.java:2098)
        at com.jogamp.opengl.GLProfile.initProfilesForDeviceCritical(GLProfile.java:1976)
        at com.jogamp.opengl.GLProfile.initProfilesForDevice(GLProfile.java:1895)
        at com.jogamp.opengl.GLProfile.initProfilesForDefaultDevices(GLProfile.java:1861)
        at com.jogamp.opengl.GLProfile.access$000(GLProfile.java:80)
        at com.jogamp.opengl.GLProfile$1.run(GLProfile.java:239)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:312)
        at com.jogamp.opengl.GLProfile.initSingleton(GLProfile.java:225)
        at com.jogamp.opengl.GLProfile.getProfileMap(GLProfile.java:2324)
        at com.jogamp.opengl.GLProfile.get(GLProfile.java:1042)
        at com.jogamp.opengl.GLProfile.get(GLProfile.java:1071)
        at com.jogamp.opengl.GLProfile.getMaxFixedFunc(GLProfile.java:824)
        at javax.media.j3d.JoglPipeline.initialize(JoglPipeline.java:131)
        at javax.media.j3d.Pipeline.createPipeline(Pipeline.java:92)
        at javax.media.j3d.MasterControl.loadLibraries(MasterControl.java:858)
        at javax.media.j3d.VirtualUniverse.<clinit>(VirtualUniverse.java:267)
        at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:194)
        at me.drton.jmavsim.Simulator.<init>(Simulator.java:193)
        at me.drton.jmavsim.Simulator.main(Simulator.java:944)
        ... 5 more`

julianoes, 2022-09-26:
I don't know if this ever worked using WSL2. These days we are focusing on ignition and gazebo classic. Support for jmavsim is no longer top of mind I'm afraid.

julianoes, 2022-09-26:
Also check https://docs.px4.io/main/en/simulation/jmavsim.html#troubleshooting