PX4/PX4-Autopilot #20303: Fixed wing control SITL in offboard mode

m1ntzz, 2022-09-25:
Hi everyone,
I have two questions.

Question 1
I would like to realize a fixed wing fleet algorithm with Mavros and Gazebo. The leader follows a virtual point, and the followers follow the leader.
In algorithm, the followers’ velocity needs to change with the velocity of leader. But according to PX4 user guide, fixed wing can receive the setpoint form position, attitude, body rate and thrust. Velocity setpoint and acceleration setpoint are automatically ignored. Is there a way to control the fixed wing linear velocity?

Question 2
I try to realize the part of leader and virtual point. In algorithm, virtual point’s velocity changes with leader’s velocity and the distance of these two. And leader’s yaw rate changes with relative position of these two.
I set the height in position setpoint, I set the yaw rate in body rate setpoint and I set the thrust in thrust setpoint. In PX4 user guide, attitude and body rate must be set with thrust but I don’t know why.
The result is: When it switches to offboard, plane will climb to expected height quickly. Once heading changed, plane would lose height and crash finally.
I can’t control it even fly along a circle. Please help!!

Jaeyoung-Lim, 2022-09-26:
> Question 1
I would like to realize a fixed wing fleet algorithm with Mavros and Gazebo. The leader follows a virtual point, and the followers follow the leader.
In algorithm, the followers’ velocity needs to change with the velocity of leader. But according to PX4 user guide, fixed wing can receive the setpoint form position, attitude, body rate and thrust. Velocity setpoint and acceleration setpoint are automatically ignored. Is there a way to control the fixed wing linear velocity?

This is currently not supported. There is limited support for using velocity setpoints to control the direction of the vehicle, but not the ground speed. you can change the speed of the vehicle using `MAV_CMD_DO_CHANGE_SPEED` message (https://mavlink.io/en/messages/common.html#MAV_CMD_DO_CHANGE_SPEED). However, this would be atomic, and not continuous.

> Question 2
I try to realize the part of leader and virtual point. In algorithm, virtual point’s velocity changes with leader’s velocity and the distance of these two. And leader’s yaw rate changes with relative position of these two.
I set the height in position setpoint, I set the yaw rate in body rate setpoint and I set the thrust in thrust setpoint. In PX4 user guide, attitude and body rate must be set with thrust but I don’t know why.
The result is: When it switches to offboard, plane will climb to expected height quickly. Once heading changed, plane would lose height and crash finally.

If you are setting body rate or attitude setpoint, you need to take care of the flight controls to keep the vehicle in the air. This is why you need to also set the thrust values

m1ntzz, 2022-09-27:
@Jaeyoung-Lim 
Thanks!! It helps a lot.
Do you know how to estimate the proper thrust values set?

Jaeyoung-Lim, 2022-09-30:
@m1ntzz The easiest way is to measure it on your system

closed 2022-10-04 by Jaeyoung-Lim