PX4/PX4-Autopilot #20322: ROS2 Waypoint Push capability

farhangnaderi, 2022-09-28:
## Feature requested
As a chat we had during the Dev-Call I opened this issue for requesting what is needed to be ported from MAVROS Fast-DDS RTPS. 

There are currently two PRs that I thought might be related to the current issue:

https://github.com/PX4/PX4-Autopilot/pull/17094
https://github.com/PX4/PX4-Autopilot/pull/16739

## Preferred Solution

For now the priority is to bring this Waypoint push to Fast RTPS instead of using OFFBOARD mode and passing a chain of points since handling OFFBOARD is a bit higher level sometimes. 

## Alternatives
1- For now MAVROS has Waypoint push capability that could be the alternative if the user wants to rely completely on ROS echo-system. 
2- he other handy solution indeed is by MAVLINK. 

It would be awesome if I am given some guides in case I can dig in this a bit more. 

CC: @dagar

dagar, 2022-10-04:
Can you comment on your experience with mission handling with MAVROS? Is pushing/pulling the entire list of waypoints at once desirable or should we break it up a bit?

[image]

farhangnaderi, 2022-10-09:
I think at the first stage pushing pulling would work. This will also shorten the time to have it at the very first stage. What I basically do in Mavros is to push a list containing all the information in each WaypointPush message. Then Takeoff and change mode to mission.