PX4/PX4-Autopilot #20334: Camera trigger not working with control allocation

Jaeyoung-Lim, 2022-09-30:
## Describe the bug
I have tried setting up the camera trigger with the control allocation, but I could not get it to work

I am using a pixhawk4 and trying to trigger a Sony A7R camera using the seagul UAV Sync2

## Expected behavior
The camera triggers on command

## To reproduce
Run `camera_trigger_test` in `nsh` or try to command trigger with QGC

## Log Files and Screenshots
- Flight log: https://review.px4.io/plot_app?log=79032fc0-a74c-49ae-872e-f7e43db14428
- Setup of control allocation
[image]

## Additional context
- I

Jaeyoung-Lim, 2022-10-04:
Camera capture seems to be not enabled even if `CAM_CAP_FBACK` is `Enabled`

```
INFO  [camera_capture] Capture enabled : NO
INFO  [camera_capture] Frame sequence : 0
INFO  [camera_capture] No trigger yet
INFO  [camera_capture] Number of overflows : 0
INFO  [camera_capture] Status chan: 1 edges: 0 last time: 0 last state: 0 overflows: 0 latency: 0
```

julianoes, 2022-10-05:
I just ran into this as well trying to set up capture.

julianoes, 2022-10-05:
From what I can tell, the capture pin is already configured as a PWM output here:
https://github.com/PX4/PX4-Autopilot/blob/06a0aedbdbd717cecc79b83a4686985950d83019/platforms/nuttx/src/px4/stm/stm32_common/io_pins/io_timer.c#L444

bkueng, 2022-10-05:
@julianoes @Jaeyoung-Lim this broke recently by reordering startup commands. Can you test https://github.com/PX4/PX4-Autopilot/pull/20362?

Jaeyoung-Lim, 2022-10-05:
I also noticed that when the `TRIG_INTERFACE` is set to Seagull MAP2, the trigger pin does not output the correct pwm output. The workaround is to set `TRIG_INTERFACE` to Generic PWM

hamishwillee, 2022-11-08:
@Jaeyoung-Lim Doesn't SEAGULL require _two triggers_? http://docs.px4.io/main/en/peripherals/camera.html#trigger-output-pin-configuration - I only see one in your actuator setup.

IN any case, when you (and/or @julianoes) get this setup with camera capture and trigger working, can you please 
1. reverify the docs are accurate
2. document your setup? Image of the wiring, details of the trigger settings, and image of the actuator screen. 
   - I can probably write something if you can detail your settings.

If you look at what is there now I have updated it in discussion with Beat about a month ago. But the core of this, including the geotagging stuff is more than 6 years old. The camera capture stuff is entirely "anecdotal". What we really need is stuff that people can copy-paste.

Make sense?

Jaeyoung-Lim, 2022-11-08:
@saengphet Unfortunately I was not able to get it to work on a pixhawk4

@hamishwillee Once I get it to work yes :smile: 

The Seagull can be used with two wires, but this is for the "keep alive" function and is not required

junwoo091400, 2022-12-02:
For visibility: Thread in discord where we discussed this as well: https://discord.com/channels/1022170275984457759/1042960930977353820/1042961591240495134

closed 2024-08-02 by Jaeyoung-Lim