PX4/PX4-Autopilot #20345: DShot not working on Kakute H7 on latest main

j-chen-opteran, 2022-10-03:
## Describe the bug
My Kakute H7 board is unable to spin up the motors on the drone, if I compile firmware using commit c9441bb on the master branch. It appears that the old DSHOT_CONFIG parameter has been replaced with PWM_MAIN_FUNCx and PWM_MAIN_TIMx. 

Reverting to Stable release v1.13.0 eliminates the issue. I can see DSHOT_CONFIG again, and setting it to any value apart from 0 lets me spin up the motors. 

## To Reproduce
Steps to reproduce the behavior:
1. Build firmware using latest master
2. Do basic setup, and test the motors in the Actuators tab

## Expected behavior
Setting PWM_MAIN_FUNCx and PWM_MAIN_TIMx correctly allows motors to be spun up.

On older firmware `dshot status` in the Nuttx shell gives the following:

```
NuttShell (NSH) NuttX-10.2.0
nsh> dshot status
INFO  [dshot] Outputs initialized: yes
INFO  [dshot] Outputs used: 0x3ff
INFO  [dshot] Outputs on: yes
dshot: cycle: 53011 events, 1050783us elapsed, 19.82us avg, min 2us max 103us 4.399us rms
INFO  [mixer_module] Param prefix: PWM_MAIN
control latency: 39180 events, 10857704us elapsed, 277.12us avg, min 262us max 2743us 144.482us rms
INFO  [mixer_module] Switched to rate_ctrl work queue
INFO  [mixer_module] Mixer loaded: yes
INFO  [mixer_module] Driver instance: 0
Channel Configuration:
Channel 1: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
Channel 2: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
Channel 3: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
Channel 0: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
Channel 4: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
Channel 5: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
Channel 6: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
Channel 7: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
Channel 8: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
Channel 9: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
Channel 10: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
Channel 11: value: 0, failsafe: 0, disarmed: 0, min: 109, max: 1999
```

On current firmware the following section is missing from the output:

```
INFO  [mixer_module] Mixer loaded: yes
INFO  [mixer_module] Driver instance: 0
```

## Drone (please complete the following information):
- Custom 3" build

bkueng, 2022-10-04:
The implementation and parameters changed, can you follow https://docs.px4.io/main/en/config/actuators.html for the setup?

j-chen-opteran, 2022-10-04:
@bkueng Thanks for the quick response. I've attempted following the new setup guide, however I experienced the following:
- Identify & assign motors is not working even after after I set PWM_MAIN_TIM0 and TIM1 to a DShot protocol. Clicking the button doesn't make them spin.
- The Actuator Testing sliders don't work either. 
- I can't see a setting to switch between MAIN and AUX PWM outputs for driving the motors.
My Kakute H7 is a V1.2 if that helps.

bkueng, 2022-10-05:
Did you reboot? Do the ESC's initialize? What is the output of `dshot status`?

julianoes, 2023-01-06:
In my case, while testing KakuteH7v2, `vehicle_angular_velocity` was not started which meant that dshot output was not frequent enough and therefore the ESCs didn't initialize.

This fixed it for me: https://github.com/PX4/PX4-Autopilot/pull/20873

closed 2023-01-06 by julianoes