PX4/PX4-Autopilot #20349: Potential divide by zero in estimator_interface.cpp

spiderkeys, 2022-10-04:
I encountered a scenario with one of our vehicle frames in sim where a division by zero occurs at:
https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/ekf2/EKF/estimator_interface.cpp#L88
wherein `_obs_buffer_length` has the value of `1`, thus 0 on the right side of the division.

On this specific vehicle frame, we only use IMU data and external vision data for all observations (baro, mag, etc all disabled). If the `_params.ev_delay_ms` value is small (say 10ms or less), at the end of this initialization block, the final `max_time_delay_ms` value ends up being 10ms and the `filter_update_period_ms` is also 10ms.
https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/ekf2/EKF/estimator_interface.cpp#L471-L501

As a result of these two values being 10ms, the `_imu_buffer_length` becomes `1`:
https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/ekf2/EKF/estimator_interface.cpp#L506

And the final `_obs_buffer_length` becomes `1` because `_imu_buffer_length` is the min value:
https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/ekf2/EKF/estimator_interface.cpp#L515

I'm not sure what the best approach to resolving this case is. A minimum observation buffer length of 2 would solve it, but I'm not sure if this would be correct.

spiderkeys, 2022-10-04:
Testing on hardware, I found that the divide by zero was also happening there, but was not being detected or causing an exception (at least on our H743 board).

Re-examining the code here:
https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/ekf2/EKF/estimator_interface.cpp#L88
I'm wondering why the subtraction of 1 is being performed. Is it actually necessary?

junwoo091400, 2022-12-02:
Interesting find!

dagar, 2023-01-05:
I believe this should be fixed with https://github.com/PX4/PX4-Autopilot/commit/84d1435880ed21a89a30466694f251e5b2bbaca9.