PX4/PX4-Autopilot #20367: DSHOT Telemetry in FW

BAI-djohnson, 2022-10-05:
I am attempting to receive DSHOT telemetry in my custom fixed wing for RPM and temperature. My system works if I select a MC so I know the wiring is correct but it is not working in FW airframe.

- PX4 1.13.0 Fixed wing airframe
- Cube Blue, standard carrier

DSHOT 1200 works correctly and I am able to send commands, but I am not receiving any telemetry back.  When I run dshot status, I get no successful frames and only timeouts.  Exact same wiring setup works with airframe selected as MC.

Any insight on how to get this working would be much appreciated.

dagar, 2022-10-06:
I can't think of a reason it would be different between MC & FW.

Do you have logs or anything you can share between the working and non-working setup?

BAI-djohnson, 2022-10-06:
I just remade the logs to make them smaller and do a true one to one comparison.  Please look at these instead.

Not working in FW with DSHOT telemetry: https://review.px4.io/plot_app?log=0ed87edf-3ed0-4c9e-b7d6-9fa163f33937
Working in MC with DSHOT telemetry: https://review.px4.io/plot_app?log=6a9c611d-662d-40ed-b988-3b791f07bca8

BAI-djohnson, 2022-10-07:
[image]
This appears to be where the issue is located.  
Located in dshot.cpp:255
I am not sure how to fix this without hard coding it.

BAI-djohnson, 2022-10-07:
I also know this is not a correct solution but I was able to get logging of esc data by forcing the logger to include the topic by changing this to add_topic instead of add_optional_topic
[image]

bkueng, 2022-10-11:
@BAI-djohnson can you try with latest `main`? This will let you use the new control allocation interface, which should give you the right amount of motors. Docs for that are here: https://docs.px4.io/main/en/config/actuators.html.

BAI-djohnson, 2022-10-11:
> @BAI-djohnson can you try with latest `main`? This will let you use the new control allocation interface, which should give you the right amount of motors. Docs for that are here: https://docs.px4.io/main/en/config/actuators.html.

Yes.  The second set of log files I uploaded were down with the current main and used the actuators tab for configuration.  Same issue, but I will attempt again today and report back if I have different results.