PX4/PX4-Autopilot #20374: PX4 SITL failing to connect to gazebo for boat model/world

altmattr, 2022-10-07:
## Describe the bug
`make px4_sitl gazebo` is working for most possible worlds and models that I try.  However, for `make px4_sitl gazebo_boat` it hangs at

~~~
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
PX4 SIM HOST: localhost
INFO  [simulator_mavlink] Waiting for simulator to accept connection on TCP port 4560
~~~

I'm running in WSL on windows 10
## To Reproduce
Steps to reproduce the behavior - in WSL command prompt:
1. `>git clone https://github.com/PX4/PX4-Autopilot.git --recursive`
2. `> ./PX4-Autopilot/Tools/setup/ubuntu.sh`
3. `HEADLESS=1 make px4_sitl gazebo_boat`
4. See error

## Expected behavior
Expected PX4 SITL start up to continue.

## Additional context
I've tried verbose logging of gzserver but no clues.  I've tried all other gazebo worlds and models and they all work, including uuv_hippocampus.

altmattr, 2022-10-10:
I have tested on ubuntu running direct on hardware as well - same behaviour

altmattr, 2022-10-11:
Bug does not exist in v1.13.0.  If I checkout that branch and follow the same steps, the boat simulation works.

altmattr, 2022-10-11:
bug _does_ occur on tag `pr-rover-ratesp`

Jaeyoung-Lim, 2022-10-16:
@altmattr This was fixed in https://github.com/PX4/PX4-SITL_gazebo/pull/917

closed 2022-10-17 by Jaeyoung-Lim