PX4/PX4-Autopilot #20379: BMI270 PX4 Driver Support

vincentpoont2, 2022-10-08:
BMI270 is a common IMU used in many flight controller such as the Kakute H7 v2. 
It would be great PX4 can support this IMU.

vincentpoont2, 2022-10-12:
@dagar Would you be able to support with this?

dagar, 2022-10-12:
I have a branch together, but it's not quite done. I'll get a pull request started.

matthewoots, 2022-10-13:
Would love to test this out too, I have a board with a bmi270 and I can help with any test :)

shupx, 2022-10-20:
I also found that kakute h7 with mpu6000 was discontinued. The only version I can buy is the kakute h7 v2 with a BMI270 imu, which is not supported by PX4. 
The Holybro website also confirmed that the support of PX4 on kakute h7 v2 is in progress. [https://docs.holybro.com/fpv-flight-controller/kakute-h7-v2/supported-firmware](https://docs.holybro.com/fpv-flight-controller/kakute-h7-v2/supported-firmware)
Hope for quick support for kakute h7, which might be the most powerful fight controller of FPV.

RENyunfan, 2022-11-12:
I just wonder if there is any progress for the support of BMI270, the kakute
 H7 v2 is so impressive. I have the hardware and willing to help with any test.

dev10110, 2022-11-19:
Hi, I am interested in getting the h7 v2 as well. please let me know what I can do to help get the sensor supported

shupx, 2022-11-19:
@dev10110 @RENyunfan @matthewoots There is a pull request about BMI270 driver on kakute mini v1.3 developed by [j-chen-opteran](https://github.com/j-chen-opteran). But he has some problem with the auto start of this imu. You can try his branch:https://github.com/PX4/PX4-Autopilot/pull/20545