PX4/PX4-Autopilot #20395: Tiltrotor VTOL: unallcated yaw torque

sfuhrer, 2022-10-12:
## Describe the bug
There seems to be very often some left over (unallocated) yaw torque, even though the tilts are not saturating. 

## To Reproduce
I guess it's reproducible in SITL, didn't try yet.

## Expected behavior
No unallocated yaw torque unless actuators are really saturating (in this case were clearly not), and thus also don't have a frozen integrator half of the flight. 

## Log Files and Screenshots
[image]

Something similar (at least similar symptoms) I fixed recently for fixed-wing control surfaces: https://github.com/PX4/PX4-Autopilot/pull/20346

sfuhrer, 2022-10-12:
@bkueng curious for your input here. 

Cleanest solution is probably to either move to a customized allocation for tiltrotors (at least handle the saturation similarly to what is done for helicopters), or make the desaturation less [multicopter-specific](https://github.com/PX4/PX4-Autopilot/blob/bdd043f27f80040f142639acfa2df7cdd63d15ef/src/modules/control_allocator/ControlAllocation/ControlAllocationSequentialDesaturation.cpp#L227). 
For a short term fix I would propose to parametrize [the thresholds for saturation detection](https://github.com/PX4/PX4-Autopilot/blob/bdd043f27f80040f142639acfa2df7cdd63d15ef/src/modules/mc_rate_control/MulticopterRateControl.cpp#L224), then that could be temporarily increased for tiltrotors.

bkueng, 2022-10-13:
It seems to me that we shouldn't be using the unallocated torque for this directly. There's cases where the smallest amount of unallocated torque saturates already, while in others, as here, there's more room.
I'd rather let the allocation publish saturation values in the fixed range of [-1, 1], which allows the rate controller to gradually reduce, and at the same time the allocation can set the values to {-1, 0, 1} in the general case, and add more granularity where we know better.
And then use a more appropriate desaturation logic for tiltrotors.

sfuhrer, 2022-11-02:
Closing as https://github.com/PX4/PX4-Autopilot/pull/20518 is in.

closed 2022-11-02 by sfuhrer