PX4/PX4-Autopilot #20409: make micrortps_agent error

mengchaoheng, 2022-10-14:
## Describe the bug
in `PX4-Autopilot/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_agent/build`, run make, have error. In the generated code is Timesync_Publisher.h but the file is named timesync_Publisher.h

## To Reproduce
Steps to reproduce the behavior:
1. on a new raspberry pi 4 run `make px4_sitl gazebo` everything is ok after set up a PX4 development environment on Ubuntu-mate 20.04. The branch is main.
2. Uploaded mission '....' (attach QGC mission file)
3. install Fast RTPS(DDS) 2.0.2 (or later) and Fast-RTPS-Gen 1.0.4 (not later!) 
4. make px4_sitl_rtps
5. cd `PX4-Autopilot/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_agent/build` run `cmake ..` and `make`
6. see error

## Expected behavior
A clear and concise description of what you expected to happen.
```Console
mch@mch-desktop:~/PX4-Autopilot/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_agent/build$ cmake ..
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenSSL: /usr/lib/aarch64-linux-gnu/libcrypto.so (found version "1.1.1f")  
-- Performing Test SUPPORTS_CXX14
-- Performing Test SUPPORTS_CXX14 - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mch/PX4-Autopilot/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_agent/build
mch@mch-desktop:~/PX4-Autopilot/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_agent/build$ make
Scanning dependencies of target micrortps_agent
[  1%] Building CXX object CMakeFiles/micrortps_agent.dir/src/RtpsTopics.cpp.o
In file included from /home/mch/PX4-Autopilot/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_agent/src/RtpsTopics.h:39,
                 from /home/mch/PX4-Autopilot/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_agent/src/RtpsTopics.cpp:34:
/home/mch/PX4-Autopilot/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_agent/src/microRTPS_timesync.h:46:10: fatal error: Timesync_Publisher.h: No such file or directory
   46 | #include "Timesync_Publisher.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/micrortps_agent.dir/build.make:63: CMakeFiles/micrortps_agent.dir/src/RtpsTopics.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
mch@mch-desktop:~/PX4-Autopilot/build/px4_sitl_rtps/src/modules/micrortps_bridge//micrortps_agent/build$/micrortps_agent/build$ cd /home/mch/PX4-Autopilot/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_agent/src
mch@mch-desktop:~/PX4-Autopilot/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_agent/src$ find Timesync_Publisher.h
find: ‘Timesync_Publisher.h’: No such file or directory
mch@mch-desktop:~/PX4-Autopilot/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_agent/src$ find timesync_Publisher.h
timesync_Publisher.h
mch@mch-desktop:~/PX4-Autopilot/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_agent/src$ 

```

mengchaoheng, 2022-10-14:
```Console
mch@mch-desktop:~/PX4-Autopilot$ git checkout e780a583
Updating files: 100% (1021/1021), done.
Note: switching to 'e780a583'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at e780a583cd Fix typo follow height: FLW_TGT_HT
mch@mch-desktop:~/PX4-Autopilot$ git submodule update --init --recursive 
Submodule 'Tools/flightgear_bridge' (https://github.com/PX4/PX4-FlightGear-Bridge.git) registered for path 'Tools/simulation/flightgear/flightgear_bridge'
Submodule 'Tools/simulation/gazebo/sitl_gazebo' (https://github.com/PX4/PX4-SITL_gazebo.git) registered for path 'Tools/simulation/gazebo/sitl_gazebo'
Submodule 'Tools/simulation/jmavsim/jMAVSim' (https://github.com/PX4/jMAVSim.git) registered for path 'Tools/simulation/jmavsim/jMAVSim'
Submodule 'Tools/simulation/jsbsim/jsbsim_bridge' (https://github.com/PX4/px4-jsbsim-bridge.git) registered for path 'Tools/simulation/jsbsim/jsbsim_bridge'
Submodule 'platforms/nuttx/NuttX/apps' (https://github.com/PX4/NuttX-apps.git) registered for path 'platforms/nuttx/NuttX/apps'
Submodule 'platforms/nuttx/NuttX/nuttx' (https://github.com/PX4/NuttX.git) registered for path 'platforms/nuttx/NuttX/nuttx'
Submodule 'src/drivers/cyphal/legacy_data_types' (https://github.com/PX4/public_regulated_data_types.git) registered for path 'src/drivers/cyphal/legacy_data_types'
Submodule 'src/drivers/cyphal/libcanard' (https://github.com/opencyphal/libcanard.git) registered for path 'src/drivers/cyphal/libcanard'
Submodule 'src/drivers/cyphal/public_regulated_data_types' (https://github.com/opencyphal/public_regulated_data_types.git) registered for path 'src/drivers/cyphal/public_regulated_data_types'
Submodule 'src/drivers/gps/devices' (https://github.com/PX4/PX4-GPSDrivers.git) registered for path 'src/drivers/gps/devices'
Submodule 'src/drivers/uavcan/libuavcan' (https://github.com/dronecan/libuavcan.git) registered for path 'src/drivers/uavcan/libuavcan'
Submodule 'src/lib/crypto/libtomcrypt' (https://github.com/PX4/libtomcrypt.git) registered for path 'src/lib/crypto/libtomcrypt'
Submodule 'src/lib/crypto/libtommath' (https://github.com/PX4/libtommath.git) registered for path 'src/lib/crypto/libtommath'
Submodule 'src/lib/crypto/monocypher' (https://github.com/PX4/Monocypher.git) registered for path 'src/lib/crypto/monocypher'
Submodule 'src/lib/events/libevents' (https://github.com/mavlink/libevents.git) registered for path 'src/lib/events/libevents'
Submodule 'src/modules/mavlink/mavlink' (https://github.com/mavlink/mavlink.git) registered for path 'src/modules/mavlink/mavlink'
Submodule 'src/modules/microdds_client/Micro-XRCE-DDS-Client' (https://github.com/eProsima/Micro-XRCE-DDS-Client.git) registered for path 'src/modules/microdds_client/Micro-XRCE-DDS-Client'
Submodule 'src/modules/micrortps_bridge/micro-CDR' (https://github.com/PX4/Micro-CDR.git) registered for path 'src/modules/micrortps_bridge/micro-CDR'
Submodule path 'Tools/simulation/flightgear/flightgear_bridge': checked out 'f47ce7b5fbbb3aa43d33d2be1f6cd3746b13d5bf'
Submodule path 'Tools/simulation/flightgear/flightgear_bridge/models/Rascal': checked out 'cbdb2fc7dd44331da1dcd04a891ed7f8c27b1ef0'
Submodule path 'Tools/simulation/flightgear/flightgear_bridge/models/TF-G1': checked out 'fdb6dd685f55097230c0eb29aa7e9f5b8d0abf1e'
Submodule path 'Tools/simulation/flightgear/flightgear_bridge/models/TF-G2': checked out '99529413531106836f4303c5389928a4fef04b05'
Submodule path 'Tools/simulation/flightgear/flightgear_bridge/models/TF-R1': checked out '158e429a0110589b1e2adfe868e166271470e144'
Submodule path 'Tools/simulation/gazebo/sitl_gazebo': checked out '07552959bc49bb1333e17a0ecd29ddd2ff70829c'
Submodule path 'Tools/simulation/gazebo/sitl_gazebo/external/OpticalFlow': checked out '28ef45c1fcb532b2bfd54c16b059c6a545143b2f'
Submodule path 'Tools/simulation/gazebo/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker': checked out '9b5b567ae77f93d1ef4d36151529ba2506a46e7a'
Submodule path 'Tools/simulation/jmavsim/jMAVSim': checked out '66b764ada522893c05224950aa6268c809f8e48a'
Submodule path 'Tools/simulation/jmavsim/jMAVSim/jMAVlib': checked out 'c08380767b21938391bec676a43b36b837af9ba4'
Submodule path 'Tools/simulation/jsbsim/jsbsim_bridge': checked out '68de2cc63ded9a0d6641d45e9eb3ed2b43454cba'
Submodule path 'Tools/simulation/jsbsim/jsbsim_bridge/models/ATI-Resolution': checked out '40571cd808a204f4062128daeb6bb6db8c134c48'
Submodule path 'Tools/simulation/jsbsim/jsbsim_bridge/models/Rascal': checked out 'd0d1db1961549e557885002cf9c3cff40cc27be2'
Submodule path 'platforms/nuttx/NuttX/apps': checked out '6d9010f47e7b00d9d86bbdd4630c78cd8d57ba35'
Submodule path 'platforms/nuttx/NuttX/nuttx': checked out '85bd35b8b906dcd87425107ff6ab2119b2c0d451'
Submodule path 'src/drivers/cyphal/legacy_data_types': checked out '36a01e428b110ff84c8babe5b65667b5e3037d5e'
Submodule path 'src/drivers/cyphal/libcanard': checked out 'db87ea32aa092c48ea103963138b6346dd3e9008'
[... cut ...]
-- Installing: /home/mch/PX4-Autopilot/build/px4_sitl_rtps/src/modules/microdds_client/share/microcdr/cmake/microcdrTargets.cmake
-- Installing: /home/mch/PX4-Autopilot/build/px4_sitl_rtps/src/modules/microdds_client/share/microcdr/cmake/microcdrConfig.cmake
-- Installing: /home/mch/PX4-Autopilot/build/px4_sitl_rtps/src/modules/microdds_client/share/microcdr/cmake/microcdrConfigVersion.cmake
[916/916] Linking CXX shared library src/examples/dyn_hello/examples__dyn_hello.px4mod
mch@mch-desktop:~/PX4-Autopilot$ ^C
mch@mch-desktop:~/PX4-Autopilot$ 

```

but the same version on another computer is ok.

mengchaoheng, 2022-10-14:
@dagar

beniaminopozzan, 2022-10-15:
@mengchaoheng , if `microRTPS_timesync.h` is trying to include `Timesync_Publisher.h` it means that in its template: `microRTPS_timesync.h.em` the `ros2_distro` variable has been defined : https://github.com/beniaminopozzan/PX4-Autopilot/blob/3c290d812d27e38445f99c20e031434d0941da86/msg/templates/urtps/microRTPS_timesync.h.em#L69-L79

This means that `generate_microRTPS_bridge.py` was called with the `ros2_distro` environment variable defined https://github.com/beniaminopozzan/PX4-Autopilot/blob/458e5a6b0e85661229b751c49a5fc19fd3e0aa9e/msg/tools/generate_microRTPS_bridge.py#L194-L200

That's definitively what you want when you are generating and compiling the agent in px4_ros_com, but you don't want it when generating and compiling the agent from PX4-Autopilot.

Try building px4_sitl_rtps without having your ros2 environment sourced.

mengchaoheng, 2022-10-15:
> @mengchaoheng , if `microRTPS_timesync.h` is trying to include `Timesync_Publisher.h` it means that in its template: `microRTPS_timesync.h.em` the `ros2_distro` variable has been defined : [...]

Why not specify this in the user manual?

mengchaoheng, 2022-10-19:
> building px4_sitl_rtps without having your ros2 environment sourced

> Try building px4_sitl_rtps without having your ros2 environment sourced.

how to do that? I have installed ROS2 on my computer, how to building px4_sitl_rtps without having my ros2 environment? @beniaminopozzan

amefabris, 2022-10-24:
I tried to build without sourcing ROS2 but it doesn't work either because after you install  Fast RTPS(DDS) 2.0.2 (as suggested in the documentation in case you don't use ROS2) I get the same error. I am able to get the the micrortps_agent running only by buildng it from [here](https://github.com/PX4/micrortps_agent), but in this case, if I run the gazebo simulation from PX4-Autopilot I get the following warnings:
```
[   micrortps_agent   ]	Unexpected topic ID '16' to publish. Please make sure the agent is capable of parsing the message associated to the topic ID '16'
[   micrortps_agent   ]	Unexpected topic ID '27' to publish. Please make sure the agent is capable of parsing the message associated to the topic ID '27'
```
As a result, I am not able to see any topics from ros2 from another terminal since ros2 topic list doesn't show anything.