PX4/PX4-Autopilot #20426: Regarding the ignition

zinuok, 2022-10-17:
Hello, 
I'm trying to use the PX4 SITL with ignition.

But the ignition is not included in your Make file.
The last pull Merged log related to the ignition is [merge px4_sitl_ign into px4_sitl_default #20188](https://github.com/PX4/PX4-Autopilot/pull/20188). 
It says that  the px4_sitl_ign has been merged into the px4_sitl_default, 
but I can't find anything related to the ignition in the makefile, for the build command like:
```
make px4_sitl_default ignition
```

Are there any things I've missed?

I'm using
```
- Ubuntu: 20.04
- ROS: ROS2 foxy
- Ignition: Edifice
```

Thank you

dagar, 2022-10-17:
Can you try this?

``` Console
make px4_sitl_default ign_x500
```

zinuok, 2022-10-18:
@dagar that command outputs:
```
[0/2] Re-checking globbed directories...
ninja: error: unknown target 'ign_x500', did you mean 'gz_x500'?
make: *** [Makefile:232: px4_sitl_default] Error 1
```bash

jmackay2, 2022-10-18:
You should follow the suggestion of the output. This works for me:
```
make px4_sitl gz_x500
```

dagar, 2022-10-18:
Sorry for the confusion, you were caught up in development iteration.

It's now `gz` which to be consistent with the naming the Gazebo project has settled on.

``` Console
make px4_sitl gz_x500
```

closed 2022-10-20 by dagar