PX4/PX4-Autopilot #20437: Resource not found: px4 ROS path [0] ..

JomondLim, 2022-10-18:
After following the offboard control example (python), when running roslaunch offboard_py start_offb.launch, i received this error despite adding the extra few lines to the bashrc as instructed in the tutorial:
Resource not found: px4
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/ubuntu/catkin_ws/src/mavlink
ROS path [2]=/home/ubuntu/catkin_ws/src/mavros/libmavconn
ROS path [3]=/home/ubuntu/catkin_ws/src/mavros/mavros_msgs
ROS path [4]=/home/ubuntu/catkin_ws/src/mavros/mavros
ROS path [5]=/home/ubuntu/catkin_ws/src/mavros/mavros_extras
ROS path [6]=/home/ubuntu/catkin_ws/src/offboard_py
ROS path [7]=/home/ubuntu/catkin_ws/src/mavros/test_mavros
ROS path [8]=/opt/ros/melodic/share
The traceback for the exception was written to the log file

This is what my .bashrc file is looking like:

source /opt/ros/melodic/setup.bash
source ~/PX4-Autopilot/Tools/simulation/gazebo/setup_gazebo.bash ~/PX4-Autopilot ~/PX4-Autopilot/build/px4_sitl_default
source /home/ubuntu/catkin_ws/devel/setup.bash
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/PX4-Autopilot
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:~/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/usr/lib/aarch64-linux-gnu/gazebo-9/plugins

Would appreciate if someone helps with this issue.
Thanks in advance

Jaeyoung-Lim, 2022-10-18:
@JomondLim Are you sure your PX4 autopilot source exists in the path `/PX4-Autopilot` ?

You can check with
```
ls /PX4-Autopilot
```

JomondLim, 2022-10-19:
@Jaeyoung-Lim 

Hi thanks for the reply, 

When entering ls /PX4-Autopilot in home directory
ls: cannot access '/PX4-Autopilot': No such file or directory

stevenchangg, 2022-10-20:
Hi @JomondLim

Currently I am facing the same issue as you :( 
Have u figured out the solution to it ? Would mean a lot as I have been struggling with this issue for a week  @Jaeyoung-Lim 
Thank you

FrancSerra, 2023-03-08:
I'm currently facing the exact same problem. DId you guys managed to fix this? I really need help on this. Thank you very much

beniaminopozzan, 2023-03-09:
@FrancSerra are you referring to this script?
https://docs.px4.io/main/en/ros/mavros_offboard_python.html#launching-your-script

FrancSerra, 2023-03-12:
> @FrancSerra are you referring to this script? https://docs.px4.io/main/en/ros/mavros_offboard_python.html#launching-your-script

Yes indeed I am

beniaminopozzan, 2023-03-12:
ok, then try to use
```sh
source ~/PX4-Autopilot/Tools/simulation/gazebo-classic/setup_gazebo.bash ~/PX4-Autopilot ~/PX4-Autopilot/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/PX4-Autopilot
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
```
instead of
```sh
source ~/PX4-Autopilot/Tools/simulation/gazebo/setup_gazebo.bash ~/PX4-Autopilot ~/PX4-Autopilot/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/PX4-Autopilot
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins
```

the reason is the `sitl_gazebo` has been renamed `sitl_gazebo-classic` and it is now located in `Tools/simulation/gazebo-classic/sitl_gazebo-classic`

pranavc2255, 2023-04-07:
> ok, then try to use
> 
> ```shell
> source ~/PX4-Autopilot/Tools/simulation/gazebo-classic/setup_gazebo.bash ~/PX4-Autopilot ~/PX4-Autopilot/build/px4_sitl_default
> export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/PX4-Autopilot
> export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
> ```
> 
> instead of
> 
> ```shell
> source ~/PX4-Autopilot/Tools/simulation/gazebo/setup_gazebo.bash ~/PX4-Autopilot ~/PX4-Autopilot/build/px4_sitl_default
> export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/PX4-Autopilot
> export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/PX4-Autopilot/Tools/simulation/gazebo/sitl_gazebo
> export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins
> ```
> 
> the reason is the `sitl_gazebo` has been renamed `sitl_gazebo-classic` and it is now located in `Tools/simulation/gazebo-classic/sitl_gazebo-classic`

Did anyone solve this issue?

I am still getting the same error, even changing to gazebo-classic?

pranavc2255, 2023-04-07:
> I'm currently facing the exact same problem. DId you guys managed to fix this? @stevenchangg @JomondLim. I really need help on this. Thank you very much

Did you solve my friend?

Jaeyoung-Lim, 2023-04-08:
@pranavc2255 High chance that your change to `gazebo-classic` was incomplete.
Please share what you actually did

mgrallos, 2023-07-04:
I'm getting the same error and I'm new to PX4. what if I use this for simulatio "make -j8 px4_sitl_default gazebo "
 and not gazebo-classic? 
Also, I don't see a "simulation" folder inside my /PX4-Autopilot/Tools directory.

beniaminopozzan, 2023-07-05:
@mgrallos 
please share which PX4 version / commit / tag you are currently using.

mgrallos, 2023-07-07:
> @mgrallos please share which PX4 version / commit / tag you are currently using.
it says v1.12.3

[image]

mgrallos, 2023-07-07:
here's the inside my PX4-Autopilot/Tools directory 
[image]

beniaminopozzan, 2023-07-07:
@mgrallos , you are working with version 1.12 then, please follow the gazebo simulation documentation for that version
https://docs.px4.io/v1.12/en/simulation/gazebo.html

closed 2024-07-10 by Jaeyoung-Lim