PX4/PX4-Autopilot #20454: release v1.13.1 error "vehicle_command lost, generation xxx -> xxx"

473867143, 2022-10-20:
## Describe the bug
Mavros disconnects when running offboard node,Shut down the node mavros to restore the connection.System console prompt "ERROR [mavlink] COMMAND_LONG vehicle_command lost, generation 1496 -> 2259" ,
"ERROR [commander] vehicle_command lost, generation 2273 -> 2276",
"ERROR [navigator] vehicle_command lost, generation 2266 -> 2365",
"ERROR [mavlink] vehicle_command_ack lost, generation 839 -> 879"

[image]

When I use the main branch or realease v1.12.3 branch, Don't have this problem.

473867143, 2022-10-20:
solved it .

closed 2022-10-20 by 473867143

SriKrishnaTKVSS, 2023-01-30:
how @473867143 ? Can you explain?