PX4/PX4-Autopilot #20463: Where can I find the files for the model if750a that are listed in the px4 user guide?

tgkang43, 2022-10-21:
Dear

http://docs.px4.io/main/en/simulation/gazebo.html
It's mentioned in the link above, " make px4_sitl gazebo_if750a ".
No matter how much I search, I can't find the relevant files and how to run it.
I am desperate for help from experts.

Best Regards,

Jaeyoung-Lim, 2022-10-21:
@tgkang43 IF750a is no longer supported and has been removed in https://github.com/PX4/PX4-SITL_gazebo/pull/891

tgkang43, 2022-10-21:
Sorry, 
I'd appreciate it if you could tell me more specifically how to find the model if750a files.

Jaeyoung-Lim, 2022-10-21:
@tgkang43 You can open the PR link, open the "Files Changed" tab and find the removed file

tgkang43, 2022-10-21:
Thank you I got it, but what about binary files like *.stl ?

Jaeyoung-Lim, 2022-10-21:
@tgkang43 You can checkout to the commit before the file was removed: https://www.atlassian.com/git/tutorials/using-branches/git-checkout

tgkang43, 2022-10-21:
Sorry I'm a beginner.
Do you mean to do it this way?
 $ git init
 $ git https://github.com/PX4/PX4-SITL_gazebo/pull/891/files
 $ git remote add origin https://github.com/PX4/PX4-SITL_gazebo/pull/891/files
 $ git ls-files --deleted | xargs git checkout --
 ... ?
It is difficult for me without prior knowledge.
Sorry, but please help me more. 
Thank you !

tgkang43, 2022-10-21:
I looked at the history with 'github desktop', but there is nothing I can do.

[image]

tgkang43, 2022-10-26:
I can get the model files including if750a from the link below.
https://git.sjtu.edu.cn/1070037389/px4-sitl_gazebo/-/tree/c0d926f7b15126498a05c466891d08083b864afc/models
Thank you !

closed 2022-10-26 by tgkang43