PX4/PX4-Autopilot #20465: the plane just fly away home in RTL mode

kevinuav, 2022-10-21:
## Describe the bug
Hey guys.
    I am using v1.13.0 stable to fly my fixedwing. But a FC's strange behavior almost make my plane lost. The plane was in RTL mode. However it just fly away from the home point. Meanwhile the setpoint of the RTL altitude changed spontaneously. It should be a bug in this version.

## To Reproduce
Steps to reproduce the behavior:
1. the plane is flying in any mode.2. 
2.switch the mode to RTL.
3.the plane fly back home first.
4.the plane just fly away home occasionally in RTL mode even you do nothing.
## Expected behavior
the plane just fly away home occasionally in RTL mode.

## Log Files and Screenshots
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/main/en/getting_started/flight_reporting.html)).
- Upload the log to the [PX4 Flight Review](http://logs.px4.io/)
- [Share the link to the log (Copy and paste the URL of the log)](https://logs.px4.io/plot_app?log=fcf6cfd4-0669-4bf3-9f35-42869d4dbdd9)

Add screenshots to help explain your problem.
[image]

## Drone (please complete the following information):
-fixedwing 
-autopilot:CUAV's V5
## Additional context
The plane just flighted away from home point at about 6:50 in RTL mode.

tstastny, 2022-10-21:
https://github.com/PX4/PX4-Autopilot/pull/20464

sfuhrer, 2023-02-15:
this should be solved with https://github.com/PX4/PX4-Autopilot/pull/20464, please re-open if not.

closed 2023-02-15 by sfuhrer