PX4/PX4-Autopilot #20478: Constant preflight fail log messages about not being connected to ground station

chris1seto, 2022-10-25:
## Describe the bug
In head of main px4, (~ 9159f020cb4051714f56e2a1d860c407b977bc94) there is constant log spam about not being connected to a ground station. While it might be common to be connected to a ground station, it's not required.  I think this error message could better be used if it appears only once after a previously connected ground station disconnects.

## Alternative solution
Another possible solution would be a param bitfield which can be used to select which preflight checks were desired to be checked. 

## To Reproduce
Steps to reproduce the behavior:
1. Power on Px4 without ground station connected
2. Log messages of the form `WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station` will be continuously printed

## Log
https://review.px4.io/plot_app?log=d2984592-304c-46f8-94c4-9222392b2228

bkueng, 2022-10-26:
Yes some of these are a bit spamming at the moment. But note that we'll completely remove these mavlink_log's once the GCS has the events UI in a release.
For now I'm fine with reducing the verbosity a bit. Do you want to do a PR?

abdulmelikbekmez, 2023-10-12:
[image]

You can use this parameter to avoid these situations.